ros2/rclcpp #1142: Start using PIMPL extensively in user facing classes, to improve ABI stability

ivanpauno, 2020-05-28:
Feature request

#### Feature description

We have run many times in the issue of not being able to backport a fix because it would break ABI.
e.g.:

- https://github.com/ros2/rclcpp/pull/1119 is almost impossible to backport.
- https://github.com/ros2/rclcpp/pull/1056

#### Implementation considerations

We should evaluate when the effort matters and where it doesn't.

wjwwood, 2020-06-08:
I have some concern with the PIMPL pattern being used extensively, due to complexity and performance.

The complexity comes from just having so many more indirections within the code, especially in the case that you need to write a template method but access data via the impl class, which can cause some awkwards scenarios where you need to type erase data.

The performance is just chasing pointers can be expensive if done everywhere. I'd like to consider the "fast PIMPL" pattern, which is a bit odd, but it can result in lots of performance improvements in certain cases:

http://www.gotw.ca/gotw/028.htm

ivanpauno, 2020-06-09:
FastPIMPL looks good to me.

There's also a lot of places were we can take advantage of an indirection we are already doing and have ABI stability for free. e.g.:

```cpp
std::shared_ptr<Publisher<MessageT>>
create_publisher(...)
...
class Publisher : public PublisherBase
```

We are already returning a pointer to the object and never the object directly.
Because we're inheriting from [enable_shared_from_this](https://github.com/ros2/rclcpp/blob/bba9dce253d98a31a3069218504d26c4be9fb8b8/rclcpp/include/rclcpp/publisher_base.hpp#L55), we actually shouldn't construct a `Publisher` or `PublisherBase` object directly.

With some slight modifications to the current API, we can provide ABI stability:

1. Make `PublisherBase` constructors protected.
  Document that classes that inherit from `PublisherBase` must have private (or protected) constructors and a friend factory function that returns a `shared_ptr`.
2. Delete copy constructor/copy assignment of `PublisherBase` (or make them private).
3. Do 1. and 2. for `Publisher`.
4. Ensure we don't provide methods that when inlined can break ABI stability. i.e.: don't explicetly refer to members in methods that can be inlined, use getters instead.

1 2 3 will make impossible for an user to hold a `Publisher`/`PublisherBase` object directly.
They will have to hold always a `std::shared_ptr<PublisherBase>` ...

4 ensures we don't break ABI stability because of inlined methods.
  e.g. instead of accessing `intra_process_is_enabled_` [here](https://github.com/ros2/rclcpp/blob/bba9dce253d98a31a3069218504d26c4be9fb8b8/rclcpp/include/rclcpp/publisher.hpp#L212), we should write `this->is_intra_process_enabled()`.

This last rule is hard to follow correctly and needs careful reviewers ...

We could also provide `PublisherHandle` and `PublisherBaseHandle` classes, in replacement of `std::shared_ptr<Pub*>`.
That can make ABI stability a bit easier to follow.

wjwwood, 2020-06-09:
Those all seem like reasonable things to try. In general I'm in favor of going over the API and trying to find places to make use more future proof, but we should just be really thoughtful for each of them rather than blindly applying PIMPL everywhere.

ivanpauno, 2020-06-09:
> Those all seem like reasonable things to try. In general I'm in favor of going over the API and trying to find places to make use more future proof, but we should just be really thoughtful for each of them rather than blindly applying PIMPL everywhere.

Yes I agree, I changed the title so the goal is clearer.