ros2/rclcpp #1147: Topic Statistics 0 publish period publishes statistics as fast as possible

emersonknapp, 2020-05-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Fat Binary
- Version or commit hash:
  - https://ci.ros2.org/view/packaging/job/packaging_linux/1883/
- DDS implementation:
  - rmf_fastrtps_cpp
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
SubscriptionOptions options; 
options.topic_stats_options.publish_period = 0ms;
node->create_subscription(..., options);
```

#### Expected behavior

Throw an error, or don't publish, not sure which is more appropriate.

#### Actual behavior

Publishes at about 2500Hz, probably the rate of the spin loop?

#### Additional information

fujitatomoya, 2020-06-01:
How about publish just once at the destruction of this subscription?

emersonknapp, 2020-06-01:
That would also be fine - i think the most important bit is documentation that clarifies the behavior

PR #1151 (2020-06-11): Add check for invalid topic statistics publish period
Resolves #1147
closed 2020-06-11 by wjwwood