ros2/rclcpp #1154: deprecation warning not working on Linux

wjwwood, 2020-06-03:
According to reports deprecation warnings for things like `rclcpp::exectuors::FutureResultCode` are not coming up on Linux:

https://github.com/ros2/ros2_documentation/pull/740#issue-427305703

wjwwood, 2020-06-04:
So I was able to get deprecation warnings in my 20.04 VM with this diff:

```diff
diff --git a/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp b/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp
index 1e1668b..da9f086 100644
--- a/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp
+++ b/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp
@@ -144,7 +144,7 @@ public:
    *  exec.spin_until_future_complete(future);
    */
   template<typename ResponseT, typename TimeRepT = int64_t, typename TimeT = std::milli>
-  rclcpp::FutureReturnCode
+  rclcpp::executor::FutureReturnCode
   spin_until_future_complete(
     std::shared_future<ResponseT> & future,
     std::chrono::duration<TimeRepT, TimeT> timeout = std::chrono::duration<TimeRepT, TimeT>(-1))

```

```
--- stderr: rclcpp
In file included from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/static_executor_entities_collector.cpp:23:
/home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp:150:96: warning: ‘using FutureReturnCode = enum class rclcpp::FutureReturnCode’ is deprecated: use rclcpp::FutureReturnCode instead [-Wdeprecated-declarations]
  150 |     std::chrono::duration<TimeRepT, TimeT> timeout = std::chrono::duration<TimeRepT, TimeT>(-1))
      |                                                                                                ^
In file included from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executor.hpp:34,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp:26,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/static_executor_entities_collector.cpp:23:
/home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/future_return_code.hpp:48:7: note: declared here
   48 | using FutureReturnCode [[deprecated("use rclcpp::FutureReturnCode instead")]] = FutureReturnCode;
      |       ^~~~~~~~~~~~~~~~

[... many other similar instances ...]

In file included from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors.hpp:23,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/rclcpp.hpp:146,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/test/test_guard_condition.cpp:19:
/home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/static_single_threaded_executor.hpp:150:96: warning: ‘using FutureReturnCode = enum class rclcpp::FutureReturnCode’ is deprecated: use rclcpp::FutureReturnCode instead [-Wdeprecated-declarations]
  150 |     std::chrono::duration<TimeRepT, TimeT> timeout = std::chrono::duration<TimeRepT, TimeT>(-1))
      |                                                                                                ^
In file included from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executor.hpp:34,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/executors.hpp:21,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/rclcpp.hpp:146,
                 from /home/william/ros2_ws/src/ros2/rclcpp/rclcpp/test/test_guard_condition.cpp:19:
/home/william/ros2_ws/src/ros2/rclcpp/rclcpp/include/rclcpp/future_return_code.hpp:48:7: note: declared here
   48 | using FutureReturnCode [[deprecated("use rclcpp::FutureReturnCode instead")]] = FutureReturnCode;
      |       ^~~~~~~~~~~~~~~~
---
```

@maryaB-osr can you say what you were doing before? Maybe it doesn't work in some specific case like when building the example packages or some other corner case I'm not considering?

clalancette, 2020-06-04:
Sorry, I should have been more explicit.  If you follow the services tutorial https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Service-And-Client/ (in particular the "client" side), and build it, you'll get no deprecation warning on Linux, though you do get on on macOS.

ivanpauno, 2020-06-09:
It sounds like deprecating the enum (rclcpp::executor::FutureReturnCode) doesn't deprecate the enumerators (e.g. rclcpp::executor::FutureReturnCode::Success) on Linux.
I'm not sure if that can be worked around, seems to be a compiler bug.

peterpena, 2020-06-19:
Following @clalancette 's step in which you build `add_two_ints_client.cpp` , the deprecation warning does not appear. With the diff provided by @wjwwood  in which you declare the enum class `rclcpp:executor::FutureReturnCode` , it does produce a deprecation warning. It seems the deprecation only works for the enum class and not the actual enumerators as mentioned by @ivanpauno .

wjwwood, 2020-06-29:
Well that's unfortunate, but it sounds like we should just close this. We don't have a good way to expose this deprecation on GCC that I can see.

@peterpena did you have a chance to look for a gcc bug about this? Maybe they have a suggested workaround that we didn't think about.

jacobperron, 2020-10-21:
Closing this since the deprecation warning was removed in https://github.com/ros2/rclcpp/pull/1327

closed 2020-10-21 by jacobperron