ros2/rclcpp #1166: Programmatically configure log level

threeal, 2020-06-11:
## Feature Request

### Feature Description

A c++ function to set log level of a specific logger under the `rclcpp::Logger` class.

### Implementation Considerations

- the current implementation could be done using the following codes, but it's based on c, not c++:
  ```c++
  if (rcutils_logging_set_logger_level(logger.get_name(), RCUTILS_LOG_SEVERITY_INFO) != 
  RCUTILS_RET_OK)
  {
    RCUTILS_LOG_FATAL("could not set the logger level");
  }
  ```
- add a `void` function that throw an error on fail or add a `boolean` function that return false on fail under the `rclcpp::Logger` class.
- (optional) add a more c++ friendly enum to be used to set the log level instead of using the RCUTILS_LOG_SEVERITY enum.

iuhilnehc-ynos, 2020-08-26:
@threeal

Do you want to make a PR for this? If you're busy, I'd like to contribute to this feature.

threeal, 2020-08-26:
@iuhilnehc-ynos sure, you may contribute to this feature

PR #1284 (2020-09-11): Add set_level for rclcpp::Logger
Close #1166
closed 2020-09-11 by wjwwood