ros2/rclcpp #1174: count_subscribers / count_publishers do not respect remapping

mikeferguson, 2020-06-12:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, Foxy
- Installation type:
  - From debians
- Version or commit hash:
  - rclcpp 2.0.0
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I created a full test node here: https://github.com/mikeferguson/ros2_bugs/blob/master/rclcpp_issues/src/count_subscriber_issue.cpp

If you run the node without remapping:
```
ros2 run rclcpp_issues count_subscriber_issue 
```
and then echo it:
```
$ros2 topic echo test
linear:
  x: 42.0
  y: 0.0
  z: 0.0
angular:
  x: 0.0
  y: 0.0
  z: 0.0
---
```
You get the expected output:
```
$ros2 run rclcpp_issues count_subscriber_issue
[INFO] [1591928821.734257323] [count_subscriber_issue]: Num of subscribers: 0
[INFO] [1591928822.734171864] [count_subscriber_issue]: Num of subscribers: 0
[INFO] [1591928831.734110847] [count_subscriber_issue]: Num of subscribers: 1
[INFO] [1591928832.734076414] [count_subscriber_issue]: Num of subscribers: 1
[INFO] [1591928833.734027599] [count_subscriber_issue]: Num of subscribers: 1
```
However, if you remap and then echo the "remapped_test" topic:
```
ros2 run rclcpp_issues count_subscriber_issue --ros-args -r test:=remapped_test
```
The subscriber will still subscribe properly and echo the data, but the num of subscribers remains 0 forever.

#### Expected behavior

I would expect that the count would still be 1 even when remapped. In theory you'd want to define a string that is your "default topic name" and use that same string for create_publisher and count_subscribers and remapping would happen seamlessly under the hood.

#### Actual behavior

Things don't work with remapping.

fujitatomoya, 2020-06-12:
+1, this seems to be clearly bug.

mikeferguson, 2020-06-12:
As a note, it looks like the [get_subscription_count()](http://docs.ros2.org/latest/api/rclcpp/classrclcpp_1_1PublisherBase.html#a60c84260b2ff0fb364fad688ed91234f) of PublisherBase does work with remapping.

fujitatomoya, 2020-06-13:
it is because rmw is not aware of remapping but rcl. for supporting this, I believe that we need to change rcl.

fujitatomoya, 2020-07-06:
@sloretz 

I am wondering if this is bug now, since https://github.com/ros2/rclcpp/blob/84ae5a18977757b92eacab600c43d86c15cf431e/rclcpp/include/rclcpp/node.hpp#L885-L887 it returns the number of subscribers that `are advertised on a specific and given topic`, so i guess it should be agnostic from remap. could you share your thought on this?

mikeferguson, 2020-07-06:
@fujitatomoya if it is decided this is "not" a bug, I'd suggest we at least update the documentation to make this clear.

fujitatomoya, 2020-07-07:
@mikeferguson 

> I'd suggest we at least update the documentation to make this clear.

that makes sense to me.

sloretz, 2020-07-10:
> it returns the number of subscribers that are advertised on a specific and given topic, so i guess it should be agnostic from remap. could you share your thought on this?

@fujitatomoya I agree that it should not remap the topic name since it's part of the [NodeGraphInterface](https://github.com/ros2/rclcpp/blob/549988277385fcfde410e419172a874cad2413c8/rclcpp/include/rclcpp/node_interfaces/node_graph_interface.hpp#L200-L204) which uses functions [that don't consider remap rules](https://github.com/ros2/rcl/blob/944068f3d99d6a4b1130fd62e38c69b96f0e759e/rcl/include/rcl/graph.h#L538). I think their purpose is to get info about topics as they are actually named. Adding that to the `rclcpp` documentation sounds like the right fix.

sloretz, 2020-07-10:
Opened #1225 to document that graph functions in rclcpp don't apply remap rules.

PR #1225 (2020-07-10): Document graph functions don't apply remap rules
resolves #1174 by adding documentation that graph functions do not apply remap rules.
closed 2020-07-10 by sloretz