ros2/rclcpp #1175: "dynamic" service becomes unresponsive after being destroyed and created again

fferri, 2020-06-12:
## Bug report

Brief description of the use case: during the lifetime of a node we need to create and destroy a service various times. However, the service client fails to find the service the second and subsequent times it is created.

To test this, I create a node with one static (i.e. active for the whole lifetime of the node) service '/enable' of type std_srvs/srv/SetBool.

Calling '/enable' with 'data: true' will create a second service '/dynamic_service' of type std_srvs/srv/Trigger.

Calling '/enable' with 'data: false' will destroy that service.

Calling '/dynamic_service' will simply log a message "trigger called!".

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - official apt repository
- Version or commit hash:
  - foxy
- DDS implementation:
  - default

#### Steps to reproduce issue

First, in one terminal we run `./build/test_dynamic_service/test_dynamic_service`.

In another terminal:

1. `ros2 service call /enable std_srvs/srv/SetBool 'data: true'` will enable the "dynamic" service
2. `ros2 service call /dynamic_service std_srvs/srv/Trigger '{}'` will test that the dynamic service works as expected (a log message "trigger called!" will be recorded)
3. `ros2 service call /enable std_srvs/srv/SetBool 'data: false'` will disable the dynamic service
4. `ros2 service call /enable std_srvs/srv/SetBool 'data: true'` enable again
5. `ros2 service call /dynamic_service std_srvs/srv/Trigger '{}'` gets stuck at "waiting for service to become available..."

note: I might need to repeat step 5 several times until it fails, i.e. it gets stuck forever

Maybe it is enough to go through steps 1 and 2, but I tried several times and it always succeeds.

#### Code

##### src/test_dynamic_service/CMakeLists.txt
```
cmake_minimum_required(VERSION 3.5)
project(test_dynamic_service)

if(NOT CMAKE_CXX_STANDARD)
    set(CMAKE_CXX_STANDARD 14)
endif()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(std_srvs REQUIRED)

add_executable(test_dynamic_service src/test_dynamic_service.cpp)
ament_target_dependencies(test_dynamic_service rclcpp std_srvs)

install(
    TARGETS test_dynamic_service
    DESTINATION lib/${PROJECT_NAME}
)

ament_package()
```

##### src/test_dynamic_service/package.xml
```
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
  <name>test_dynamic_service</name>
  <version>0.0.0</version>
  <description>Testing creating a dynamic service</description>
  <maintainer email="federico.ferri.it@gmail.com">Federico Ferri</maintainer>
  <license>BSD2</license>

  <buildtool_depend>ament_cmake_ros</buildtool_depend>

  <depend>std_srvs</depend>

  <test_depend>ament_lint_auto</test_depend>
  <test_depend>ament_lint_common</test_depend>

  <export>
    <build_type>ament_cmake</build_type>
  </export>
</package>
```

##### src/test_dynamic_service/src/test_dynamic_service.cpp
```
#include <cstdlib>
#include <functional>
#include <rclcpp/rclcpp.hpp>
#include <std_srvs/srv/set_bool.hpp>
#include <std_srvs/srv/trigger.hpp>

using namespace std::placeholders;
using namespace std::chrono_literals;

using namespace std_srvs::srv;

rclcpp::Node::SharedPtr g_node = nullptr;
rclcpp::Service<Trigger>::SharedPtr g_dynsrv = nullptr;

void handle_dynamic_service(const std::shared_ptr<rmw_request_id_t> request_header, const std::shared_ptr<Trigger::Request> request, const std::shared_ptr<Trigger::Response> response)
{
    RCLCPP_INFO(g_node->get_logger(), "trigger called!");
    response->success = true;
}

void handle_enable_service(const std::shared_ptr<rmw_request_id_t> request_header, const std::shared_ptr<SetBool::Request> request, const std::shared_ptr<SetBool::Response> response)
{
    if(request->data)
    {
        if(g_dynsrv)
        {
            RCLCPP_INFO(g_node->get_logger(), "already enabled");
            response->success = false;
            response->message = "already enabled";
        }
        else
        {
            RCLCPP_INFO(g_node->get_logger(), "enabling dynamic service");
            g_dynsrv = g_node->create_service<Trigger>("dynamic_service", handle_dynamic_service);
            response->success = true;
        }
    }
    else
    {
        if(g_dynsrv)
        {
            RCLCPP_INFO(g_node->get_logger(), "disabling dynamic service");
            g_dynsrv = nullptr;
            response->success = true;
        }
        else
        {
            RCLCPP_INFO(g_node->get_logger(), "already disabled");
            response->success = false;
            response->message = "already disabled";
        }
    }
}

int main(int argc, char ** argv)
{
    rclcpp::init(argc, argv);
    g_node = rclcpp::Node::make_shared("test_dynamic_service");
    auto enable_server = g_node->create_service<SetBool>("enable", handle_enable_service);
    rclcpp::spin(g_node);
    rclcpp::shutdown();
    g_node = nullptr;
    return 0;
}
```

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

fferri, 2020-06-12:
I get a similar problem with a "dynamic" publisher: after it is destroyed and created again, I'm unable to receive published messages anymore.

Is there some problem in removing services / publishers while keeping the node running?

fferri, 2020-06-22:
To simplify testing, after running the node, you can use this command:

`for ((;;)); do ros2 service call /enable std_srvs/srv/SetBool 'data: true'; sleep 1; ros2 service call /dynamic_service std_srvs/srv/Trigger '{}'; sleep 1; ros2 service call /enable std_srvs/srv/SetBool 'data: false'; sleep 1; done`

and wait for it to be stuck at "waiting for service to become available...".
Today I tested again and it took 15 iterations to reach that point.

fujitatomoya, 2020-06-24:
This could be related to https://github.com/ros2/rmw_fastrtps/issues/392, i confirmed the problem can be observed with provided sample program. And using cyclonedds with `export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp`, I cannot reproduce the problem.

fujitatomoya, 2020-06-24:
CC: @richiware

richiware, 2020-07-02:
@miguelcompany could also help in this

MiguelCompany, 2020-07-20:
@fujitatomoya @fferri This should have been solved by eProsima/Fast-DDS#1295

jacobperron, 2020-08-06:
eProsima/Fast-DDS#1295 has been released into Foxy.

@fferri, can you confirm if this bug still exists? If so, feel free to re-open this ticket and leave a comment.

closed 2020-08-06 by jacobperron