ros2/rclcpp #1177: Double free in create_wall_timer if an invalid period is passed

brawner, 2020-06-16:
In [`create_wall_timer()` in `create_timer.hpp`](https://github.com/ros2/rclcpp/blob/master/rclcpp/include/rclcpp/create_timer.hpp#L105-L110), if a bad period duration is passed (either < 0 or > nanoseconds::max()) then rcl will throw an error. During the destructor of WallTimer, it will call cancel, which results in a double free. I haven't found exactly where.

This issue is to track the several PRs I think will be necessary to resolve the cascade of errors present.

brawner, 2020-06-16:
Add PR to check arguments in `create_wall_timer`: https://github.com/ros2/rclcpp/pull/1178

brawner, 2020-06-16:
For reference, this is the error message:
```
[20.871s] 8: [ RUN      ] TestCreateTimer.call_wall_timer_with_bad_arguments
[20.874s] 8: [ERROR] [1592350214.053122631] [rclcpp]: Couldn't initialize rcl timer handle: timer period must be non-negative, at /home/brawner/workspace/ros2_master/src/ros2/rcl/rcl/src/rcl/timer.c:139
[20.874s] 8: 
[20.874s] 8: terminate called after throwing an instance of 'std::runtime_error'
[20.874s] 8:   what():  Couldn't cancel timer: timer is invalid, at /home/brawner/workspace/ros2_master/src/ros2/rcl/rcl/src/rcl/timer.c:379
[20.938s] 8: -- run_test.py: return code -6
```

fujitatomoya, 2020-06-17:
@brawner 

> During the destructor of WallTimer, it will call cancel, which results in a double free.

i do not think of `doube free` case either, but exception in destructor.

checking argument is also better, but we could change like `only issue cancel on initialized rcl_timer_t successfully`, i think that would be less exception.

fujitatomoya, 2020-06-17:
problem confirmed,

```
# strace ./build/ros2_test_prover/ros2_644
...<snip>
write(1, "Callback: Getting logger", 24Callback: Getting logger) = 24
write(1, "\n", 1
)                       = 1
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x38} ---
+++ killed by SIGSEGV (core dumped) +++
```

brawner, 2020-06-17:
Agreed @fujitatomoya. This was the intended second PR to followup and resolve the issue. It throws an exception in the constructor if `rcl_timer_init` fails.

https://github.com/ros2/rclcpp/pull/1179

brawner, 2020-06-29:
I believe this has been adequately addressed by #1178 and #1179.

closed 2020-06-29 by brawner