ros2/rclcpp #1186: Inconsistent namespaces for memory strategies

brawner, 2020-06-18:
It looks like the namespacing of memory strategies is not consistent and could be due for cleanup. Please let me know if this is intentional and should be left alone.

```
// include/rclcpp/strategies
rclcpp::strategies::message_pool_memory_strategy::MessagePoolMemoryStrategy;
rclcpp::memory_strategies::allocator_memory_strategy::AllocatorMemoryStrategy;

// include/rclcpp
rclcpp::memory_strategy::MemoryStrategy;
rclcpp::memory_strategies::create_default_strategy();
```

I'm ticketing this while it's fresh in my mind, and I plan to come back around and propose a PR.

gftrobots, 2021-04-16:
Is this issue still open? The ros2_control_node indicates in the source code that there is a dependency to this issue.

https://github.com/ros-controls/ros2_control/blob/master/controller_manager/src/ros2_control_node.cpp

clalancette, 2021-04-16:
This one is still open, but this is not the issue that is referenced there.  That code references #1168 , which is still open.  That being said, we've fixed a lot of issues recently with the MultiThreadedExecutor and timers, so it would be worthwhile to retest the original problem and see if it is fixed.