ros2/rclcpp #1219: StaticSingleThreadedExecutor's spin_some, spin_all and spin_once cannot use the base class implementation

brawner, 2020-07-09:
I've been expanding the unit tests of the executors  for all three classes, and these methods in `StaticSingleThreadedExecutor` should not be using the base class implementation in `rclcpp::Executor`. Because `StaticSingleThreadedExecutor` derives and implements its own `add_node` and `remove_node`, the collection of nodes in `rclcpp::Executor` is not updated and so these spin variants won't do any work. Either `add_node` and `remove_node` need to call the base class implementations as well, which would currently throw an "This node already has an executor" exception or these variants need to be derived specifically for `StaticSingleThreadedExecutor`.

dirk-thomas, 2020-07-30:
@brawner Can you comment what you think should happen on this ticket: assign it to you, assign it to someone else, enhancement for backlog, ...?

brawner, 2020-07-30:
It's not exactly an enhancement, after dealing with the backport in #1229, I'll assign it to someone else.

fujitatomoya, 2024-11-04:
i think we can close this by https://github.com/ros2/rclcpp/pull/1338. CC: @mauropasse

closed 2024-11-04 by fujitatomoya