ros2/rclcpp #1239: this->get_clock()->now() when using sim time updates in slow blocks

Michael-Equi, 2020-07-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - rclcpp version 2.0.2 and ROS2 Foxy
- DDS implementation:
  - Cyclone
- Client library (if applicable):
  - rclcpp 

#### Steps to reproduce issue
Run an rclcpp node that gets the time using `this->get_clock()->now()` at a frequency greater than 10Hz with a gazebo simulation running at 0.9-1.0 real time factor. `use_sim_time` should be set to true. Compute the difference between last time and the current time and print to terminal.

#### Expected behavior
The time should be interpolated automatically so as to have more fine updates that are in line with the real time factor of the simulation.

#### Actual behavior
Time updates at only 10Hz and results in an output like something below where the larger number is the current time output by `this->get_clock()->now()`.seconds() (.nanoseconds() has the same problem) and the smaller number is the difference between the last time a callback was called and the current time. This is used in my program to compute the velocity of an object but the low resolution results in mostly `nans` being output. The callback in my program runs at around 200Hz so the low resolution of the time update is much more exaggerated in my case. Without `use_sim_time=True` the behavior is as expected and the time updates fast enough to provide accurate velocities, although they are not in line with the real time factor of the simulation. 

```
959.668
0
959.668
0
959.668
0
959.668
0
959.668
0
959.668
0
959.668
0
959.768
0.1
959.768
0
959.768
0
959.768
0
959.768
0
959.768
0
959.768 
```

clalancette, 2020-08-04:
My understanding on how this works is that the time is published via a [Gazebo ROS plugin](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/e1de3c14d089e4588d7b3ffd0d03453403effdbb/gazebo_ros/src/gazebo_ros_init.cpp#L133).  In turn, that publication (I believe) is tied to the world update loop inside of Gazebo.  So I don't think this directly has to do with rclcpp, though I could be wrong.  Pinging @chapulina @scpeters for some advice here (or at least a pointer to who we should contact).

chapulina, 2020-08-04:
> that publication (I believe) is tied to the world update loop inside of Gazebo

Yup, it is.

> 0.9-1.0 real time factor

Besides the RTF, it's also important to note the update rate. By default, Gazebo is updated at 1000 Hz, so I'd expect that to be the frequency of new messages being published.

> 959.668

Are those print outs of the seconds? I'm not sure what the `0` mean.

> The time should be interpolated automatically

I don't think that's a good idea. You can do extra interpolation on top of the time you're receiving, but when using simulation, I think it's important to get the exact time that your simulation thinks it's in, even if it's not as granular as you'd like.

Michael-Equi, 2020-08-04:
@chapulina Those are seconds. The zeros are from when now - last time = 0. The node I am using this for is supposed to use the difference between the read times of the current and last message published at 200Hz in order to compute wheel velocity but whenever now = last time this causes division by zero. The world is running at 1000Hz as noted in my world file `     <real_time_update_rate>1000</real_time_update_rate>` but the time does not seem to be updating at 1000Hz, it looks more like 10Hz.

chapulina, 2020-08-04:
Ok, I remember now that the default clock frequency is 10 Hz. You can test locally with:

```
$ ros2 launch gazebo_ros gazebo.launch.py &
$ ros2 topic hz /clock
average rate: 9.967
	min: 0.100s max: 0.101s std dev: 0.00027s window: 11
average rate: 9.971
	min: 0.100s max: 0.101s std dev: 0.00022s window: 21
average rate: 9.970
	min: 0.100s max: 0.101s std dev: 0.00021s window: 31
average rate: 9.972
	min: 0.100s max: 0.101s std dev: 0.00019s window: 41
average rate: 9.971
	min: 0.100s max: 0.101s std dev: 0.00021s window: 51

```

You can change that with the [publish_rate](https://github.com/ros-simulation/gazebo_ros_pkgs/blob/e1de3c14d089e4588d7b3ffd0d03453403effdbb/gazebo_ros/src/gazebo_ros_init.cpp#L141) parameter.

It would be great to have this documented on a tutorial [here](http://gazebosim.org/tutorials?cat=connect_ros).

clalancette, 2020-08-20:
My understanding here is that this is not a bug, though the tutorials could be updated.  As such, I'm going to close this out here, but please feel free to open an issue/PR to the Gazebo tutorials to fix this up.  Thanks.

closed 2020-08-20 by clalancette