ros2/rclcpp #1255: [BUG] Namespacing the parameter blackboard node leads to wrong events topic

maxlein, 2020-08-03:
## Bug report

Our system has all nodes namespaced, e.g. /robot_1/param_server.  
Now we switched from eloquent to foxy, and all parameter events failed to fire.  

After a bit of testing and asking a [question](https://answers.ros.org/question/358170/parameter-events-on-foxy/?comment=358296#post-id-358296), I found out that adding a namespace breaks parameter events. 

Output from demo parameter blackboard started with namespace `test_ns`:  

```
ros2 node info /test_ns/parameter_blackboard
/test_ns/parameter_blackboard
  Subscribers:
    /test_ns/parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
  Service Servers:
    /test_ns/parameter_blackboard/describe_parameters: rcl_interfaces/srv/DescribeParameters
    /test_ns/parameter_blackboard/get_parameter_types: rcl_interfaces/srv/GetParameterTypes
    /test_ns/parameter_blackboard/get_parameters: rcl_interfaces/srv/GetParameters
    /test_ns/parameter_blackboard/list_parameters: rcl_interfaces/srv/ListParameters
    /test_ns/parameter_blackboard/set_parameters: rcl_interfaces/srv/SetParameters
    /test_ns/parameter_blackboard/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically
```  

parameter_events node:  

```
ros2 node info /test_ns/parameter_events
/test_ns/parameter_events
  Subscribers:
    /test_ns/parameter_events: rcl_interfaces/msg/ParameterEvent
  Publishers:
    /parameter_events: rcl_interfaces/msg/ParameterEvent
    /rosout: rcl_interfaces/msg/Log
```

As you see the parameter_events topic subscriber has a namespace, whereas the publisher has not.  
This doesn't look right to me..  

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binaries, foxy
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
Start parameter blackboard node and parameter events node from demo_cpp project with a namespace, e.g. add --ros-args -r __ns:=/test_ns

#### Expected behavior
Parameter events are handled the same as without a namespace

#### Actual behavior
No parameter events are called

fujitatomoya, 2020-08-04:
@maxlein 
CC: @clalancette 

I confirmed that problem's fixed, but it would be nice if you could double check. then we can close this.

ivanpauno, 2020-08-04:
Fixed in https://github.com/ros2/rclcpp/pull/1257.

@maxlein If it doesn't work for you, let me know and I will reopen (the bug was corrected in master, it hasn't been fixed in Foxy yet).

closed 2020-08-04 by ivanpauno