ros2/rclcpp #1265: support wildcards for just namespaces, or just nodes

rpaaron, 2020-08-09:
## Feature request

#### Feature description
support wildcards for just namespaces, or just nodes

```yaml
/**:
    ros__parameters:
        applies_to_all_nodes_in_all_namespaces: true
```

would be nice also to have
```yaml
some_namespace:
    *:
        ros__parameters:
            applies_to_all_nodes_in_particular_namespace: true
```
and
```yaml
/*:
    some_node:
        ros__parameters:
            applies_to_particular_node_in_all_namespaces: true
```

is there a way currently to achieve this?

fujitatomoya, 2020-08-24:
I think that it would be really nice to describe wildcard format in https://index.ros.org/doc/ros2/Tutorials/Node-arguments/#setting-parameters-from-yaml-files. (After this PR is merged.)

iuhilnehc-ynos, 2020-08-25:
@rpaaron

Let me confirm the feature as following,

- The following configuration can be accepted by a node on 'some_namespace' namespace, whatever the node name is.

```
some_namespace:
    *:
        ros__parameters:
            applies_to_all_nodes_in_particular_namespace: true
```

- The following configuration can be accepted by a node name 'some_node' on whatever the namespace is.

```
/*:
    some_node:
        ros__parameters:
            applies_to_particular_node_in_all_namespaces: true
```

If the maintainer here accepts these features, I'd like to contribute to it.

iuhilnehc-ynos, 2020-08-25:
@rpaaron 

Sorry, I didn't see your PR. Delete my previous comment.

clalancette, 2020-09-10:
@rpaaron Thanks for the issue and the pull request implementing it.  Before we go too far on the implementation, though, we should settle on the design first.  Towards that end, could you please propose a modification to http://design.ros2.org/articles/static_remapping.html for the design?  Thanks.

jacobperron, 2020-10-09:
Correct me if I'm wrong, but using wildcards as proposed here looks consistent with how [name matching is described in the design doc](http://design.ros2.org/articles/static_remapping.html#match-part-of-a-rule). It still might be useful to add concrete examples when it comes to using them in parameter files, either in the design doc or in the tutorial as @fujitatomoya suggests.

rpaaron, 2020-10-12:
I think these changes are inline with what's written in the design document already.  If not, could someone please advise what needs to change.

jacobperron, 2020-10-12:
It wasn't obvious to me where the best place to document this feature is, but I think I've found an appropriate place and opened a PR: https://github.com/ros2/design/pull/303

Reviews are appreciated :slightly_smiling_face:

AlexKaravaev, 2021-12-16:
Are there any plans on merging this? Or maybe achieving same functionality with latest rolling distrib. 

I think this feature is quite helpful

fujitatomoya, 2022-02-04:
https://github.com/ros2/rclcpp/pull/1839 should also fix this problem.

fujitatomoya, 2022-02-04:
@AlexKaravaev it would be really nice if you could check that https://github.com/ros2/rclcpp/pull/1839 fixes your problem.

tonynajjar, 2023-07-04:
This should be solved right?

tonynajjar, 2023-07-04:
On the other hand, I noticed that this is not fixed for Composable Nodes. Any idea why? Is this a bug in https://github.com/ros2/rclcpp/pull/1839 or a missing feature?

fujitatomoya, 2023-07-05:
@iuhilnehc-ynos can you check https://github.com/ros2/rclcpp/issues/1265#issuecomment-1620208946?

iuhilnehc-ynos, 2023-07-13:
Please see https://github.com/ros2/rclcpp/issues/715#issuecomment-767539717.

iuhilnehc-ynos, 2023-07-13:
`ComposableNode` supported both formats in https://github.com/ros2/launch_ros/pull/259.
I think we need a follow-up PR to support wildcards to support ComposableNode.