ros2/rclcpp #1273: topic statistic tests failing with Connext

dirk-thomas, 2020-08-11:
In the nightly jobs using only Connext these test fail since they have been added in #1080:

* http://build.ros2.org/view/Rci/job/Rci__nightly-connext_ubuntu_focal_amd64/43/testReport/rclcpp/TestSubscriptionTopicStatisticsFixture/test_receive_stats_for_message_no_header/
* http://build.ros2.org/view/Rci/job/Rci__nightly-connext_ubuntu_focal_amd64/43/testReport/rclcpp/TestSubscriptionTopicStatisticsFixture/test_receive_stats_for_message_with_header/

@prajakta-gokhale @dabonnie Will either of you be able to work on this this in the near term?

dabonnie, 2020-08-11:
I can take a look (next week), but it looks like you beat us to it :-)

dirk-thomas, 2020-08-12:
> I can take a look (next week)

It would still be good to take another look at the test. While my PR makes it pass with Connext I am not sure why the original test assumed every metrics message has not-zero samples.

> but it looks like you beat us to

Please consider reviewing the pending pull request if you have a chance.

dabonnie, 2020-08-12:
@dirk-thomas will review by EOD.

PR #1274 (2020-08-13): fix topic stats test, wait for more messages, only check the ones with samples
Fixes #1273.

Before: [image]
After: [image]

@prajakta-gokhale @dabonnie FYI
closed 2020-08-13 by dirk-thomas