ros2/rclcpp #1282: test_executors test failing on Windows debug

dirk-thomas, 2020-08-20:
See https://ci.ros2.org/view/nightly/job/nightly_win_deb/1712/testReport/junit/projectroot/test/test_executors/ for the first failing build.

It seems this regressions was introduced in #1270.

brawner, 2020-08-20:
Thanks for bringing this to my attention. This might be related to what I was discussing in #1230, but it looks like it's more repeatable than what I was discovering and more targeted to individual tests.

brawner, 2020-08-20:
Having trouble reproducing on bare-metal. While my docker image is building, I'm also running this CI job:

[image]

brawner, 2020-08-25:
I'm still working on this, but I have to take a break to address other things. I'll come back to this soon.

chapulina, 2020-11-06:
This test has been failing consistently for a long time. I'm suggesting disabling it in https://github.com/ros2/rclcpp/pull/1443

brawner, 2020-11-24:
This was resolved by #1404 (rolling) and #1401 (foxy).

closed 2020-11-24 by brawner