ros2/rclcpp #1285: rclcpp_action server could cause deadlock with its wrapper's mutex

daisukes, 2020-08-26:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Foxy release src build 
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/4109e26bf96d0a0a9ef755364dabc851556f3089
- DDS implementation:
  - tried with Fast-RTPS, cyclonedds
- Client library (if applicable):
  - Navigation 2

#### Steps to reproduce issue

The original issue is here and I think this is rclcpp issue
https://github.com/ros-planning/navigation2/issues/1961

```
- launch tb3 simulation and make it active
- repeat sending goal action and waiting for the result in a single loop, and wait for the deadlock 
https://gist.github.com/daisukes/46298b083a48e5db7016f3c0efba7e2d
  - I could reproduce the deadlock with this code usually in a few minutes
```

#### Expected behavior

- does not stop, keep working

#### Actual behavior

- the server will not accept goals when the deadlock happens

#### Problem

- user callbacks are called in the `reentlant_mutex_` lock context
  - in the callbacks, the wrapper will lock its mutex

- in another thread, the wrapper first lock its mutex
  - then calls rclcpp_action method, which tries to lock `reentlant_mutex_`

This inconsistent locking order cause deadlocks. 
So I think rclcpp_action needs to call user callbacks outside of `reentlant_mutex_` lock context.

#### Additional information
Here is gdb information
Thread 9 and 14 are trying to hold two mutex in different order
https://gist.github.com/daisukes/712316a97832f5d9ab851ad47c77ad98

Thread 9: server thread
Frame# 6 [nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::handle_goal](https://github.com/ros-planning/navigation2/blob/2e0697c64c265f31bd4ef6c5ed43374841242077/nav2_util/include/nav2_util/simple_action_server.hpp#L82)
Frame# 14 [rclcpp_action::ServerBase::execute_goal_request_received()](https://github.com/ros2/rclcpp/blob/4a5eed968ca6fa4979214db8ac945fb163919326/rclcpp_action/src/server.cpp#L213)

Thread 14: working thread [started here](https://github.com/CMU-cabot/navigation2/blob/db7cea3e5e/nav2_util/include/nav2_util/simple_action_server.hpp#L128)
Frame# 6 [rclcpp_action::ServerBase::notify_goal_terminal_state()](https://github.com/ros2/rclcpp/blob/4a5eed968ca6fa4979214db8ac945fb163919326/rclcpp_action/src/server.cpp#L545)
Frame# 11 [nav2_util::SimpleActionServer<nav2_msgs::action::ComputePathToPose, rclcpp::Node>::succeeded_current](https://github.com/ros-planning/navigation2/blob/2e0697c64c265f31bd4ef6c5ed43374841242077/nav2_util/include/nav2_util/simple_action_server.hpp#L305)

SteveMacenski, 2020-08-26:
And to be clear, this isn't just a specific issue with Navigation2, any users that have mutex locks on their resources will run into this problem. We typically don't run into this issue because we don't in quick succession call an action server immediately after returning a result.

If you were to use actions as workers though, you'd absolutely run into this issue more frequently.

SteveMacenski, 2020-09-25:
@clalancette any thoughts?

ivanpauno, 2021-01-18:
Closing as it seems to have been fixed by https://github.com/ros2/rclcpp/pull/1313 (and #1518 will fix it for Foxy, where the bug was reported), feel free to reopen if that's not the case.

closed 2021-01-18 by ivanpauno