ros2/rclcpp #1295: Client/Service might be memory leak

iuhilnehc-ynos, 2020-09-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - d66cd96f254648de5fa68f6c28a34968563294fb
- DDS implementation:
  - rmw_fastrtps_cpp
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ colcon build --cmake-args -DCMAKE_BUILD_TYPE=Debug --packages-up-to rclcpp
$ . install/setup.bash
$ valgrind --leak-check=full ./build/rclcpp/test/test_allocator_memory_strategy
```

#### Expected behavior
No memory leak

#### Actual behavior

<details><summary> memory leak log </summary>
<p>

```shell
==1269872== 
==1269872== HEAP SUMMARY:
==1269872==     in use at exit: 57,303 bytes in 167 blocks
==1269872==   total heap usage: 253,411 allocs, 253,244 frees, 94,040,442 bytes allocated
==1269872== 
==1269872== 659 (152 direct, 507 indirect) bytes in 1 blocks are definitely lost in loss record 44 of 53
==1269872==    at 0x483DE63: operator new(unsigned long) (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==1269872==    by 0x5E7FD35: rmw_create_client (rmw_client.cpp:173)
==1269872==    by 0x50AB1EF: rmw_create_client (functions.cpp:391)
==1269872==    by 0x5070849: rcl_client_init (client.c:169)
==1269872==    by 0x23C1E2: rclcpp::Client<test_msgs::srv::Empty>::Client(rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (client.hpp:219)
==1269872==    by 0x238E06: void __gnu_cxx::new_allocator<rclcpp::Client<test_msgs::srv::Empty> >::construct<rclcpp::Client<test_msgs::srv::Empty>, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(rclcpp::Client<test_msgs::srv::Empty>*, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (new_allocator.h:147)
==1269872==    by 0x236279: void std::allocator_traits<std::allocator<rclcpp::Client<test_msgs::srv::Empty> > >::construct<rclcpp::Client<test_msgs::srv::Empty>, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::allocator<rclcpp::Client<test_msgs::srv::Empty> >&, rclcpp::Client<test_msgs::srv::Empty>*, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (alloc_traits.h:484)
==1269872==    by 0x2328BE: std::_Sp_counted_ptr_inplace<rclcpp::Client<test_msgs::srv::Empty>, std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (shared_ptr_base.h:548)
==1269872==    by 0x22E4EB: std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<rclcpp::Client<test_msgs::srv::Empty>, std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(rclcpp::Client<test_msgs::srv::Empty>*&, std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Client<test_msgs::srv::Empty> > >, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (shared_ptr_base.h:679)
==1269872==    by 0x22A515: std::__shared_ptr<rclcpp::Client<test_msgs::srv::Empty>, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Client<test_msgs::srv::Empty> > >, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (shared_ptr_base.h:1344)
==1269872==    by 0x22566E: std::shared_ptr<rclcpp::Client<test_msgs::srv::Empty> >::shared_ptr<std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::_Sp_alloc_shared_tag<std::allocator<rclcpp::Client<test_msgs::srv::Empty> > >, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (shared_ptr.h:359)
==1269872==    by 0x22038E: std::shared_ptr<rclcpp::Client<test_msgs::srv::Empty> > std::allocate_shared<rclcpp::Client<test_msgs::srv::Empty>, std::allocator<rclcpp::Client<test_msgs::srv::Empty> >, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::allocator<rclcpp::Client<test_msgs::srv::Empty> > const&, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (shared_ptr.h:702)
==1269872== 
==1269872== 1,858 (432 direct, 1,426 indirect) bytes in 3 blocks are definitely lost in loss record 48 of 53
==1269872==    at 0x483D7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==1269872==    by 0x50BFF46: __default_allocate (allocator.c:37)
==1269872==    by 0x50707CA: rcl_client_init (client.c:162)
==1269872==    by 0x23C1E2: rclcpp::Client<test_msgs::srv::Empty>::Client(rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (client.hpp:219)
==1269872==    by 0x238E06: void __gnu_cxx::new_allocator<rclcpp::Client<test_msgs::srv::Empty> >::construct<rclcpp::Client<test_msgs::srv::Empty>, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(rclcpp::Client<test_msgs::srv::Empty>*, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&) (new_allocator.h:147)
==1269872==    by 0x236279: void std::allocator_traits<std::allocator<rclcpp::Client<test_msgs::srv::Empty> > >::construct<rclcpp::Client<test_msgs::srv::Empty>, rclcpp::node_interfaces::NodeBaseInterface*, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, rcl_client_options_t&>(std::allocator<rclcpp::Client<test_msgs::srv::Empty> >&, rclcpp::Client<test_msgs::srv::Empty>*, rclcpp::node_interfaces::NodeBaseInterface*&&, std::shared_ptr<rclcpp::node_interfaces::NodeGraphInterface>&, std::__cxx11::basic_string<cha [... cut ...]
https://github.com/ros2/rclcpp/blob/d66cd96f254648de5fa68f6c28a34968563294fb/rclcpp/src/rclcpp/client.cpp#L69-L70

There are two ways to solve this memory leak.

- make sure [service/client] handle stored at external is clean before calling `ClientBase::~ClientBase`.

- put share_ptr(node_handle_) inside deleter of `[service/client]_handle_`,  it will make sure [service/client] handle is destructed before the node handle
I do like this method because it will make users not to care which variable should be destroyed first if using `get_[service/client]_handle()` outside.

iuhilnehc-ynos, 2020-09-10:
Hi, if you agree on method two, I'll continue to update it.

iuhilnehc-ynos, 2020-09-10:
Let me put `client` as an example,

https://github.com/ros2/rclcpp/blob/d66cd96f254648de5fa68f6c28a34968563294fb/rclcpp/src/rclcpp/client.cpp#L43-L64

to

```
  client_handle_.reset(
    new rcl_client_t, [handle=node_handle_](rcl_client_t * client)
    {
      if (rcl_client_fini(client, handle.get()) != RCL_RET_OK) {
        RCLCPP_ERROR(
          rclcpp::get_node_logger(handle.get()).get_child("rclcpp"),
          "Error in destruction of rcl client handle: %s", rcl_get_error_string().str);
        rcl_reset_error();
      }
      delete client;
    });
  *client_handle_ = rcl_get_zero_initialized_client();
```

fujitatomoya, 2020-09-16:
@iuhilnehc-ynos 

i see what you are trying to do here.

> put share_ptr(node_handle_) inside deleter of [service/client]_handle_, it will make sure [service/client] handle is destructed before the node handle

i am not the original author so i am not sure but i believe that `weak_ptr` is intentionally used here. i guess it is used to avoid reference chain and memory leak. maybe we could wait for opinions from others.

iuhilnehc-ynos, 2020-09-17:
> i believe that weak_ptr is intentionally used here

I believed so, that's why I need somebody's confirmation.
I thought the following before, let me use a sample to explain.
```
#include <iostream>
#include <memory>
#include <string>

class A {
public:
  A() {
    std::cout << __PRETTY_FUNCTION__ << std::endl;
    a = 10;
  }
  ~A() {
    std::cout << __PRETTY_FUNCTION__ << std::endl;
  }
  void test() {
    std::cout << __PRETTY_FUNCTION__ << ":" << a << std::endl;
  }
private:
  int a;
};

class B {
public:
  B() {
    std::cout << __PRETTY_FUNCTION__ << std::endl;
  }
  ~B() {
    std::cout << __PRETTY_FUNCTION__ << std::endl;
  }
};

int main() {
  std::shared_ptr<A> a(new A);
  // if we are sure or want that B is destructed before A, the deleter of B can be wrote as following that not to use weak_ptr here
  std::shared_ptr<B> b;
  b.reset(new B, [a](B *b) {
    a->test();
    delete b;
  });
  
  // whatever the order of calling b.reset or a.reset, the destructor of A must be called after B 
  // this is what I descriped before, user does not care which variable should be called first
  a.reset();
  b.reset();
}

```

iuhilnehc-ynos, 2020-09-17:
In ros2, node_handle is created before client_handle, so I think `node_handle must be destructed after client_handle` is the correct order, which must be promised inside the library.

- client_handle can use the shared_ptr of node_handle
- if there is a statement to set shared_ptr of client_handle to node_handle, that's wrong.

PR #1296 (2020-10-05): Make sure to clean the external client/service handle.
Fixes #1295
closed 2020-10-05 by clalancette