ros2/rclcpp #1299: rclcpp memory leak check via valgrind

fujitatomoya, 2020-09-10:
## Overview

This is a meta-ticket to manage and organize rclcpp test memory leak problems with valgrind.

## Requirement

- Check all of rclcpp test executable with valgrind.
- List all the memory leak issue and address to each one of them.

## Memory Leak List

| Issue | PR | Description |
|:---:|:---:|:---:|
| https://github.com/ros2/rclcpp/issues/1295 | https://github.com/ros2/rclcpp/pull/1296 |  |
|  |  |  |
|  |  |  |

fujitatomoya, 2020-09-10:
CC: @Barry-Xu-2018 @iuhilnehc-ynos

Barry-Xu-2018, 2020-09-11:
@fujitatomoya @iuhilnehc-ynos

Thanks for creating this ticket.   

Currently, we execute rclcpp test and find below memory leak cases.  
    rcl: 511fc40ca305fe075cd53e0b97176cc7be272d69  
    rclcpp: 01d6f52e3281f6df2c191690527f9208006a5c50  
    rmw_fastrtps_cpp: bbe2ac2e3e63c145e9dabd4f073c62d672521a57  

| test executable file | memory leak | Issue | PR |  Description |
|:--|:-:|:--|:--|:--|
| ./test_allocator_memory_strategy                    | YES | #1295 | #1296 | Client/Service might be memory leak  |
| ./test_create_subscription                          | YES | #1300| #1301  | In SubscriptionTopicStatistics, avoid self dependency that not destoryed|
| ./test_executors                                    | YES | | | |
| ./test_loaned_message                               | YES | | | |
| ./test_publisher                                    | YES | | | |
| ./test_publisher_subscription_count_api             | YES | | | |
| ./test_static_executor_entities_collector           | YES | | | |
| ./test_subscription                                 | YES | | | |
| ./test_subscription_options                         | YES | | | |
| ./test_subscription_topic_statistics                | YES | | | |

fujitatomoya, 2021-01-26:
@Barry-Xu-2018 

i think everything's been addressed by now, could you confirm and help to close this one?

Barry-Xu-2018, 2021-01-26:
@fujitatomoya 

Yeah. Please close it.

closed 2021-01-26 by fujitatomoya