ros2/rclcpp #1323: rclcpp/context lacking a const method for `shutdown_reason`

Blast545, 2020-09-21:
Adding coverage tests for `context` module (#1321), we have realized there is no `const` method for the `shutdown_reason` method. See [link to the code](https://github.com/ros2/rclcpp/blob/3a4ac0ca2093d12035070443692798b0c9f9da3a/rclcpp/include/rclcpp/context.hpp#L153 )

This issue is to track the conversation [here](https://github.com/ros2/rclcpp/pull/1321#discussion_r491231948) and get some feedback.

brawner, 2020-09-21:
Link to `.cpp` (https://github.com/ros2/rclcpp/blob/3a4ac0ca2093d12035070443692798b0c9f9da3a/rclcpp/src/rclcpp/context.cpp#L289-L294). This function requires locking the `init_mutex_` which prevents it from being const. I suggested @Blast545 open this issue, just to see if others thought there might be more preferred ways of doing this. 

One possible solution would be to make `init_mutex_` and perhaps other mutexes `mutable`.

suab321321, 2021-02-21:
@clalancette should I follow @brawner and make it the `init_mutex_` mutable?

clalancette, 2021-02-22:
> @clalancette should I follow @brawner and make it the `init_mutex_` mutable?
> and open a PR

Yeah, that is the usual way to deal with this problem.  Making `init_mutex_` mutable seems totally reasonable to me.

fujitatomoya, 2021-03-20:
done with https://github.com/ros2/rclcpp/pull/1578

closed 2021-03-20 by fujitatomoya