ros2/rclcpp #1335: ROS2 image topic publisher freezes when a subscriber is initiated at a remote PC

atb033, 2020-09-25:
## Bug report

My aim was to send a video stream as a ros-topic to a remote PC (Raspberry Pi 4). To do this, I published an image stream using the `cam2image` node of [image_tools](https://github.com/ros2/demos/tree/master/image_tools) package

At the remote PC (which was connected using WiFi), when I initiated the subscriber using `showimage` node of the same package, the `cam2image` node froze as shown in this [video](https://streamable.com/yg87db). 

**Required Info:**

- Operating System:
  - Ubuntu 20.04 (in docker image: `ros:foxy-ros-base`)
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue

Install `image_tools` in both host and remote PC

```
sudo apt update
sudo apt install ros-foxy-image-tools
```

In the host PC

```
ros2 run image_tools cam2image --ros-args -p burger_mode:=true -p frequency:=10. -p reliability:=best_effort
```

In the remote PC

```
ros2 run image_tools showimage --ros-args -p show_image:=false -p reliability:=best_effort
```

#### Expected behavior

The host PC publishes at 10 Hz, and the remote PC subscribes at the same frequency.

#### Actual behavior

The publisher node freezes when the subscriber is initiated.

#### Additional information

Also, the publisher node continues to operate normally once the subscriber node is killed

fujitatomoya, 2020-10-08:
Using `cyclonedds` solve this problem.

```
> sudo apt install ros-foxy-rmw-cyclonedds-cpp
> export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp

> ros2 run image_tools cam2image --ros-args -p burger_mode:=true -p frequency:=10. -p reliability:=best_effort (Host)
> ros2 run image_tools showimage --ros-args -p show_image:=false -p reliability:=best_effort (Remote)
```

fujitatomoya, 2020-10-08:
Raspi4 source build with https://github.com/ros2/ros2/commit/ca09b31ce2851773cf722e1fe1e64b70ee651fdd, this can be observed with https://github.com/ros2/rmw_fastrtps. (https://github.com/ros2/rmw_cyclonedds does not have this problem) I think that this issue needs to be moved to https://github.com/ros2/rmw_fastrtps.

atb033, 2020-10-08:
Thank you for your response.

Yes, the performance did improve when I used `cyclonedds`. I will close this issue and move it to https://github.com/ros2/rmw_fastrtps

Also, it looks like many of my network-related problems are now being solved when I use `cyclonedds`. Do you think it is superior to the default `fastrtps`?

closed 2020-10-08 by atb033

fujitatomoya, 2020-10-08:
> it looks like many of my network-related problems are now being solved when I use

would you mind if you could share them?

I think there is always Pros/Cons and trading-off, I would say that depends on the use case.

atb033, 2020-10-13:
My code was taking unusually long to discover services/actions after I upgraded to foxy.

In my code, there is a section where I call a service that brings up a behavior-tree. Since this behavior tree has nodes that are clients to many services/actions, it initializes many service/action clients. This step was taking a long time when I was using `rmw_fastrtps_cpp`. The discovery has now become a lot faster now that we have switched to `rmw_cyclonedds_cpp`