ros2/rclcpp #1341: graph_listener error when mocking rcl_node_get_graph_guard_condition,

Blast545, 2020-09-28:
I found this problem when adding coverage tests #1330. The following test is failing in Windows (and it's flaky on MacOS):

```c++
TEST_F(TestGraphListener, error_run_graph_listener_mock_get_graph_cond) {
  auto global_context = rclcpp::contexts::get_global_default_context();
  auto graph_listener_test =
    std::make_shared<TestGraphListenerProtectedMethods>(global_context);
  graph_listener_test->mock_start_thread();
  EXPECT_NO_THROW(graph_listener_test->add_node(node_graph()));
  EXPECT_NE(nullptr, node()->get_node_graph_interface()->get_graph_event());
  auto mock_wait_set_clear = mocking_utils::patch_and_return(
    "lib:rclcpp", rcl_node_get_graph_guard_condition, nullptr);
  RCLCPP_EXPECT_THROW_EQ(
    graph_listener_test->run_protected(),
    std::runtime_error("failed to get graph guard condition: error not set"));
}
```

I haven't investigated the reason yet, I'll take a look later this week.

CursedRock17, 2023-09-28:
Since this test doesn't exist anymore and the draft linked to this issue was solved by #1404, this can probably be closed right?

clalancette, 2023-09-28:
Yeah, I'm going to close this.  It is really stale, and if we come across something like this again we'll just debug it separately.

closed 2023-09-28 by clalancette