ros2/rclcpp #1351: rclcpp::Time and rclcpp::Duration max values not near max int64_t

Blast545, 2020-09-29:
Follow up Issue to track the conversation _in https://github.com/ros2/rclcpp/pull/1347#discussion_r496920298_

Basically this is to consider a new approach to the one agreed on #538. The `max` method for Time and Duration IMO should return the max value of the representation in nanoseconds. 

Not sure if that was a constrain in the past, but the sum / substraction operators can handle properly overflow and underflow operations, so I don't see any reason to limit the max value returned by the max method. 

If this is not a desired approach, an update to the header of the function will be required, as it stated _\return the maximum representable value_ which is not the case.

Tagging @clalancette for awareness

clalancette, 2020-10-15:
@mabelzhang Make sure to read the linked issues and discussion, and talk to @Blast545 and @wjwwood about this one.  My personal opinion is that we should make `Time::max` and `Duration::max` be the `int32::max` value for both seconds and nanoseconds; that just seems consistent.

wjwwood, 2020-11-13:
I don't think the concern is about overflowing operations, I think the concern is that if you create a `Time(std::numeric_limits<int32_t>::max(), 999999999)` and then add a second, you will not overflow the time object (as the  internal `int64_t` can store it and we don't artificially cap the value) but if you tried to convert it to the ROS message type, it would not fit as it uses int32 for the seconds field.

Personally, I think limiting the value is odd, and I think changing the storage is a bad  idea too, so I think we should either change this max to return the max value we can represent, even if that wouldn't fit in the ROS message.

The long term solution, imo, is to deprecate the existing time message and move towards one with a single `int64_t`, but that's gonna be hard to do.