ros2/rclcpp #1362: rclcpp_components_register_nodes doesn't work with recursive cmake setups

andre-nguyen, 2020-09-30:
## Bug report

When `rclcpp_components_register_nodes` is called in a subdirectory instead of the top level CMakeLists.txt it fails to register the component.

**Required Info:**

- Operating System:
  -  Ubuntu 18.04
- Installation type:
  -  ROS2 dashing from binaries
- Version or commit hash:
  -  dashing
- DDS implementation:
  -  default fast-rtps
- Client library (if applicable):
  -  rclcpp but the cmake part

#### Steps to reproduce issue

Given a recursive cmake setup such as:
```
.
└── my_package/
    ├── CMakeLists.txt
    └── src/
        ├── CMakeLists.txt
        ├── mylib1/
        │   ├── CMakeLists.txt
        │   └── some_source_file.cpp
        └── mylib2/
            └── CMakeLists.txt
```
Where `my_package/CMakeLists.txt` would have something like
```cmake
add_subdirectory(src)

ament_package()
```
and `my_package/src/CMakeLists.txt`
```cmake
add_subdirectory(mylib1)
add_subdirectory(mylib2)
```
and `my_package/src/mylib1/CMakeLists.txt`
```cmake
add_library(mylib1 some_source_file.cpp)
...
rclcpp_components_register_nodes(mylib1  "mylib1::MyComponent")
```

#### Expected behavior
The component should be properly registered regardless of where `rclcpp_components_register_nodes` is called.

#### Actual behavior
This fails to register `mylib1::MyComponent` as a component and running `ros2 component types` does not show the component. The call to `rclcpp_components_register_nodes` has to be moved up to `my_packages/CMakeLists.txt` for it to work. 

#### Additional information
Just a guess here but it might have to do with the scope of the `_RCLCPP_COMPONENTS__NODES` variable which doesn't get propagated upwards outside of the subdirectory. I'm a bit unclear however how that variable gets used at the toplevel maybe with the package hook?

If this is correct then the fix should be to set `_RCLCPP_COMPONENTS__NODES` in the cmake cache.

sloretz, 2020-10-15:
There's an answer about this problem on ROS answers. In short, there's not really a good solution to it.

https://answers.ros.org/question/361289/ros2-components-registration-from-subdirectory-cmakelists-file/

Would you be willing to make a PR documenting that the register component macros can only be used by `ament_cmake` packages and must be called in the same `CMakeLists.txt` file as `ament_package()`?

* https://github.com/ros2/rclcpp/blob/master/rclcpp_components/cmake/rclcpp_components_register_node.cmake
* https://github.com/ros2/rclcpp/blob/master/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake

andre-nguyen, 2020-10-15:
Thanks, for getting back to me, what's your opinion of using the cmake cache to propagate the values upwards?

https://github.com/ros2/rclcpp/blob/018cfaa2197b94ed1517cc2860758e1b8db02bdb/rclcpp_components/cmake/rclcpp_components_register_nodes.cmake#L66-L68

to something like

```cmake
 set(_RCLCPP_COMPONENTS_${resource_index}__NODES 
  "${_RCLCPP_COMPONENTS_${resource_index}__NODES}${_arg};${_path}/$<TARGET_FILE_NAME:${target}>\n"
  CACHE STRING FORCE) 
list(APPEND _RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES ${resource_index}) 
set(_RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES
  "${_RCLCPP_COMPONENTS_PACKAGE_RESOURCE_INDICES}"
  CACHE STRING FORCE)
```

and something similar for the single target version `rclcpp_components_register_node`.

Or is this approach unsustainable in terms of maintenance?

ivanpauno, 2021-01-08:
Closing in favor of https://github.com/ament/ament_cmake/issues/227 (which IIUC is the same issue).

ivanpauno, 2021-01-08:
@andre-nguyen your proposal looks similar too https://github.com/ament/ament_cmake/issues/227#issuecomment-750501962.
It might work fine, but it's kind of an abuse of cmake cache variables (though maybe we should give it a try).

closed 2021-01-08 by ivanpauno