ros2/rclcpp #1381: Windows compile error - min, max macro compile bug

YoavFekete, 2020-10-06:
So the compilation of my code is not working as the Rhino plugins use windows.h
I managed to locally fix the problem with the following solution:

Replace all uses of std::numeric_limits<double>::max();  with 
                               (std::numeric_limits<double>::max)();   (the additional partness solve the complier confusion, same for min)

In files of Duration and Time ,  add to to of header:
#undef max
#undef min

after that all compiler well. This solution as i understand will not add strange side effects like the  #define NOMINMAX  solution

## Bug report
- Operating System  :   Windows 10, Ubuntu 16.04.
- Installation type:         binaries
- Version or commit hash:   foxy
- DDS implementation:
- Client library (if applicable):   rclcpp, 

#### Steps to reproduce issue
  Try to write a Rhino 6 plugin with ros code.

#### Expected behavior
Code should compile

#### Actual behavior
Error due to use of min, max macros

ivanpauno, 2020-10-13:
This is not an issue in `rclcpp`, but your include of `windows.h` is affecting later includes of `rclcpp` headers.

You can use `NOMINMAX` to avoid the issue (e.g. https://github.com/ros2/rcutils/pull/229).

It's also possible to prevent the issue here, see:
- https://github.com/ros2/rclcpp/pull/1053/files
- https://github.com/ros2/rclcpp/pull/1053#issuecomment-611326100

If you can open a PR implementing one of those approaches, I think that would be fine.

ivanpauno, 2021-01-08:
Closing, as the user can currently workaround the issue.
A PR modifying all uses of `min` and `max` in `rclcpp` in a way that doesn't conflicts with `windows.h` as mentioned above would still be welcomed!

closed 2021-01-08 by ivanpauno