ros2/rclcpp #1391: TestGraphListener error_run_graph_listener_destroy_context fails consistently on Windows Debug

nuclearsandwich, 2020-10-07:
## Bug report

**Required Info:**

- Operating System:
  - Windows
- Installation type:
  - Source
- Version or commit hash:
  - Introduced in #1330 
- DDS implementation:
  - N/A
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Nightly builds since #1330 merged and https://ci.ros2.org/job/ci_windows/12590/testReport/rclcpp/TestGraphListener/error_run_graph_listener_destroy_context/ all reproduce the issue consistently.

@Blast545 @brawner Since the pull request CI doesn't cover Windows Debug I expect this failure has been buried beneath our other Windows issues but it is standing out now.

brawner, 2020-10-07:
Thanks for the report. This might be caused by similar issues to https://github.com/ros2/rclcpp/issues/1282 which is also a windows debug only error.

brawner, 2020-10-12:
See #1401 for potential solution and discussion.

brawner, 2020-10-13:
Also, it looks like this issue is a duplicate of https://github.com/ros2/rclcpp/issues/1340

brawner, 2020-10-14:
PR for fix in rolling: https://github.com/ros2/rclcpp/pull/1404

brawner, 2020-11-24:
I believe this was resolved by #1404 (rolling) and #1401 (foxy)

closed 2020-11-24 by brawner