ros2/rclcpp #1403: test_subscription_topic_statistics fails with rmw_connext_cpp

brawner, 2020-10-13:
This is a reoccuring failure over at:
http://build.ros2.org/view/Rci/job/Rci__nightly-connext_ubuntu_focal_amd64/106/testReport/rclcpp/TestSubscriptionTopicStatisticsFixture/test_receive_stats_for_message_no_header/

It was probably introduced in https://github.com/ros2/rclcpp/pull/1281, which aligns with when this failure started showing up.

iuhilnehc-ynos, 2020-10-19:
I can reproduce this failure, refer to 
```
$ ros2 topic echo /statistics statistics_msgs/msg/MetricsMessage
measurement_source_name: test_sub_stats_node
metrics_source: message_age
unit: ms
window_start:
  sec: 1603101739
  nanosec: 842547404
window_stop:
  sec: 1603101740
  nanosec: 843078333
statistics:
- data_type: 1
  data: .nan
- data_type: 3
  data: .nan
- data_type: 2
  data: .nan
- data_type: 5
  data: 0.0
- data_type: 4
  data: .nan
---
measurement_source_name: test_sub_stats_node
metrics_source: message_period
unit: ms
window_start:
  sec: 1603101739
  nanosec: 842547404
window_stop:
  sec: 1603101740
  nanosec: 843078333
statistics:
- data_type: 1
  data: .nan
- data_type: 3
  data: .nan
- data_type: 2
  data: .nan
- data_type: 5
  data: 0.0
- data_type: 4
  data: .nan
---
```

I am sure it's introduced in #1281.

#1281 lost logic,
```
    if (!has_samples) {
      continue;
    }
```

ivanpauno, 2020-10-28:
@dabonnie does @iuhilnehc-ynos comment make sense to you?
Would reintroducing that logic fix the issue?

dabonnie, 2020-10-28:
I don't think so (context [here](https://github.com/ros2/rclcpp/pull/1274/files#r469641061).

For that test I'm curious why it's only failing for connext. Notice how line [243](https://github.com/ros2/rclcpp/pull/1281/files#diff-9ee0b8f919cc9341189b14d6e4b47cd9d22fb2d740b0a4e2a66dd0729be8ea9cR243) introduces a check for empty messages, which is used at line [416](https://github.com/ros2/rclcpp/pull/1281/files#diff-9ee0b8f919cc9341189b14d6e4b47cd9d22fb2d740b0a4e2a66dd0729be8ea9cR416) to assert that empty header messages are empty. IMHO the continue block is just hiding a failure, where we don't expect a message to have empty fields at that point.

clalancette, 2022-02-03:
Since we aren't using `rmw_connext_cpp` anymore, this one isn't really that relevant.  So I'll close it out.

closed 2022-02-03 by clalancette