ros2/rclcpp #1412: init_and_remove_ros_arguments parse arguments two times

iuhilnehc-ynos, 2020-10-20:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - source
- Version or commit hash:
  - 018cfaa2197b94ed1517cc2860758e1b8db02bdb
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ ros2 run demo_nodes_cpp listener --ros-args --log-level debug
```

#### Expected behavior
```
<...>
[DEBUG] [1603184568.251432332] [rcl]: Couldn't parse arg 0 (/home/sharedata/Linux/ROS2/docker_ros2_master/local_install/lib/demo_nodes_cpp/listener) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 87, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:233
<...>
```

#### Actual behavior

```
<...>
[DEBUG] [1603184568.251432332] [rcl]: Couldn't parse arg 0 (/home/sharedata/Linux/ROS2/docker_ros2_master/local_install/lib/demo_nodes_cpp/listener) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 87, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:233
<...>
[DEBUG] [1603184568.253581614] [rcl]: Couldn't parse arg 0 (/home/sharedata/Linux/ROS2/docker_ros2_master/local_install/lib/demo_nodes_cpp/listener) as a remap rule in its deprecated form. Error: Expected lexeme type (19) not found, search ended at index 87, at /home/sharedata/Linux/ROS2/docker_ros2_master/src/ros2/rcl/rcl/src/rcl/lexer_lookahead.c:233
```

#### Additional information

https://github.com/ros2/rclcpp/blob/018cfaa2197b94ed1517cc2860758e1b8db02bdb/rclcpp/src/rclcpp/utilities.cpp#L59-L67

Fix: At https://github.com/ros2/rclcpp/blob/018cfaa2197b94ed1517cc2860758e1b8db02bdb/rclcpp/src/rclcpp/utilities.cpp#L69-L70, we need to check if `get_global_default_context()->get_rcl_context()` exist, and then decide to call `rcl_parse_arguments` or not.

iuhilnehc-ynos, 2020-10-20:
@ivanpauno 

Could you confirm whether this issue should be fixed or not?

ivanpauno, 2020-10-20:
> , we need to check if get_global_default_context()->get_rcl_context() exist, and then decide to call rcl_parse_arguments or not.

I'm not sure if that helps, as context init doesn't seem to have failed.

If you have a way of avoiding to parse the arguments twice without adding much complexity, I think that would be ok.

iuhilnehc-ynos, 2020-10-21:
> I'm not sure if that helps, as context init doesn't seem to have failed.

Thank you for your response. That helped much.
I'll use another way to update this issue, and I hope that it's not much complexity for you.

PR #1415 (2020-11-02): Avoid parsing arguments twice
Fixes #1412
closed 2020-11-02 by ivanpauno