ros2/rclcpp #1417: Creating/destroying a node results in inconsistent heap allocations

brawner, 2020-10-21:
While adding performance tests (see #1411) for creating a basic rclcpp::Node, we noticed that the number of heap allocations during each iteration can vary during each call to rclcpp::Node's constructor. Even calling init/shutdown each time didn't help the variation.

With further investigation this may lead to some potential performance improvements.

Example of benchmark:
```
BENCHMARK_F(NodePerformanceTest, create_node_with_init)(benchmark::State & state)
{
  reset_heap_counters();
  for (auto _ : state) {
    state.PauseTiming();
    rclcpp::init(0, nullptr);
    state.ResumeTiming();
    
    reset_heap_counters();
    auto node = std::make_unique<rclcpp::Node>("node");
    printf("Construct heap allocations %lu\n", get_allocation_count());
    
    benchmark::DoNotOptimize(node);
    benchmark::ClobberMemory();
    
    state.PauseTiming();
    node.reset();
    rclcpp::shutdown();
    state.ResumeTiming();
  }
}
```
Result with `rmw_fastrtps_cpp`
```
Construct heap allocations 4909
Construct heap allocations 3673
Construct heap allocations 3674
Construct heap allocations 3672
Construct heap allocations 3670
Construct heap allocations 3670
Construct heap allocations 3670
Construct heap allocations 3650
Construct heap allocations 3673
Construct heap allocations 3681
Construct heap allocations 3660
```

With a similar benchmark measuring the destructor:
```
Destruct heap allocations 76
Destruct heap allocations 77
Destruct heap allocations 77
Destruct heap allocations 76
Destruct heap allocations 76
Destruct heap allocations 76
Destruct heap allocations 78
Destruct heap allocations 76
Destruct heap allocations 76
Destruct heap allocations 75
Destruct heap allocations 75
Destruct heap allocations 78
Destruct heap allocations 75
```

And `rmw_cyclonedds_cpp`
```
Construct heap allocations 5752
Construct heap allocations 4935
Construct heap allocations 4961
Construct heap allocations 4958
Construct heap allocations 4903
Construct heap allocations 4950
Construct heap allocations 4961
Construct heap allocations 4939
Construct heap allocations 4976
Construct heap allocations 4924
```

```
Destruct heap allocations 196
Destruct heap allocations 196
Destruct heap allocations 194
Destruct heap allocations 188
Destruct heap allocations 192
Destruct heap allocations 197
Destruct heap allocations 194
Destruct heap allocations 192
Destruct heap allocations 193
Destruct heap allocations 195
Destruct heap allocations 193
Destruct heap allocations 191
Destruct heap allocations 196
Destruct heap allocations 193
```

Also note worthy is that the first iteration had a significant increase in allocations, which is likely the real use case of an rclcpp user. But the caching effects will make it difficult for benchmarks to measure those initial allocations.

ivanpauno, 2020-10-22:
If instead of calling `rclcpp::init()` and `rclcpp::shutdown()` repeatedly you create a `rclcpp::Context` in the loop, the allocations should be more consistent. The reason is that the destructor of `rclcpp::Context` does more things that only calling `rclcpp::Contex::shutdown`, and part of that state might be create in the first `rclcpp::Node` creation.

> std::make_unique<rclcpp::Node>("node")

Bear in mind that creating a `rclcpp::Node` unique pointer can cause UB, due to `rclcpp::Node` [inheriting from std::enable_shared_from_this](https://github.com/ros2/rclcpp/blob/5d9db5de748e936c102352121f7910c7d64ba2ad/rclcpp/include/rclcpp/node.hpp#L74) (that wasn't  probably an issue in that example, but just in case).

ivanpauno, 2020-10-22:
Is there any actionable item in this issue?

I mean, though I would prefer heap allocations to be consistent, this doesn't sound like something that anybody is going to work in the near future.
If that's the case, I would prefer closing the issue.

brawner, 2020-10-22:
I should note that the issue still remains if you don't init/shutdown inside the loop. I was just trying to init/shutdown inside the loop because the first call to the constructor yielded about 20% more allocations than the subsequent calls. It turns out that did not make the heap allocations consistent across the first and subsequent calls.

```
BENCHMARK_F(NodePerformanceTest, create_node_with_init)(benchmark::State & state)
{
    rclcpp::init(0, nullptr);
    for (auto _ : state) {
      reset_heap_counters();
      auto node = std::make_unique<rclcpp::Node>("node");
      printf("Construct heap allocations %lu\n", get_allocation_count());
    
      benchmark::DoNotOptimize(node);
      benchmark::ClobberMemory();
    
      state.PauseTiming();
      node.reset();
      rclcpp::shutdown();
      state.ResumeTiming();
  }
  rclcpp::shutdown();
}
```

This might be investigated as part of the performance benchmark work, so I think it is worthwhile to keep open. It seems like fluctuating heap allocations suggest that not only are there potentially more allocations being done than necessary, but also that some aren't being done deterministically. If these allocations aren't being done deterministically from one loop to another, than it's possible there is undefined behavior responsible  for the fluctuating allocations.

ivanpauno, 2020-10-22:
> I should note that the issue still remains if you don't init/shutdown inside the loop. I was just trying to init/shutdown inside the loop because the first call to the constructor yielded about 20% [...]

Yeah, what I mean is that the following should avoid the big difference in the first node construction:

```cpp
BENCHMARK_F(NodePerformanceTest, create_node_with_init)(benchmark::State & state)
{
  reset_heap_counters();
  for (auto _ : state) {
    state.PauseTiming();
    rclcpp::Context context;
    context.init(0, nullptr);
    state.ResumeTiming();
    
    reset_heap_counters();
    auto node = std::make_unique<rclcpp::Node>("node");
    printf("Construct heap allocations %lu\n", get_allocation_count());
    
    benchmark::DoNotOptimize(node);
    benchmark::ClobberMemory();
    
    state.PauseTiming();
    node.reset();
    context.shutdown();
    state.ResumeTiming();
  }
}
```

ivanpauno, 2020-10-22:
> This might be investigated as part of the performance benchmark work, so I think it is worthwhile to keep open. It seems like fluctuating heap allocations suggest that not only are there potentially [...]

The difference in the first allocation I know why it's caused.
All `nodes` in the same context are sharing a DDS participant, and that participant is created in the first node construction.

About the small differences in the remaining loops, I'm not sure.

brawner, 2020-10-22:
Adjusting your example slightly to work correctly yields similar results. I'm not seeing any improvement in reduced inconsistency. This example creates a new context for each node, and shuts it down, which in turn should destroy the dds participant, correct?

```c++
BENCHMARK_F(NodePerformanceTest, create_node_with_init)(benchmark::State & state)
{
  for (auto _ : state) {
    state.PauseTiming();
    auto context = std::make_shared<rclcpp::contexts::DefaultContext>();
    context->init(0, nullptr);
    rclcpp::NodeOptions options;
    options.context(context);

    reset_heap_counters();
    state.ResumeTiming();

    auto node = std::make_shared<rclcpp::Node>("node", options);
    printf("Construct heap allocations %lu\n", get_allocation_count());

    benchmark::DoNotOptimize(node);
    benchmark::ClobberMemory();

    state.PauseTiming();
    node.reset();
    context->shutdown("");
    context.reset();
    state.ResumeTiming();
  }
}
```
rmw_cyclonedds_cpp
```
Construct heap allocations 5758
Construct heap allocations 4954
Construct heap allocations 4954
Construct heap allocations 4977
Construct heap allocations 4966
Construct heap allocations 4961
Construct heap allocations 4959
Construct heap allocations 4967
...
```

rmw_fastrtps_cpp
```
Construct heap allocations 5029
Construct heap allocations 3776
Construct heap allocations 3742
Construct heap allocations 3766
Construct heap allocations 3755
...
```

ivanpauno, 2020-10-22:
> Adjusting your example slightly to work correctly yields similar results. I'm not seeing any improvement in reduced inconsistency. This example creates a new context for each node, and shuts it down, which in turn should destroy the dds participant, correct?

Yes, that should be destroying the participant and creating a new one.

DDS vendors might have some data of static lifetime that is initialized on first use and generates allocations, e.g.: most of them have a "domain participant factory" singleton.
But I'm not sure if that's the case here.

brawner, 2020-11-13:
Based on some investigative work, I believe these inconsistencies are to be expected, and I think I figured out where.

I prototyped some code to track allocations and their locations. It was pretty rudimentary, but it identified the main source of variability (fast-rtps) and it also helped identify the locations of initial cache allocation.
 1. The main source of initialization on the first go looks like rcutils_logging_initialize, loading shared libraries (find_library_path), get_global_default_context , rosidl_typesupport_cpp::get_typesupport_handle_function
 2. It appears the main source of variability is FastRTPS use of an AsyncWriterThread. This is a class that wakes up periodically and writes whatever is pending. It wakes up on fixed duration intervals I believe so will look like a small number of fluctuations in  allocations.
However, I also found some more peculiar fluctuations in allocations beyond the AsyncWriterThread when calling rclcpp_action::create_server. I can't yet figure out the cause of this, but it's a strange one. It had to do with entities calling `rcl_node_resolve_name`.

The prototyped code is unfortunately too slow to be useful on an ongoing basis I think. It required comparing stack traces for uniqueness on each malloc/realloc call, which understandable really slowed down the code.

I'll close this as I don't expect any more action on this issue going forward.

closed 2020-11-13 by brawner