ros2/rclcpp #1431: publish message which is 200KB per 100ms，ros2 topic hz /topicName can not get 10Hz

849631323, 2020-10-30:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - 
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):

#### Steps to reproduce issue
#include <chrono>
#include <functional>
#include <memory>
#include <string>

#include "rclcpp/rclcpp.hpp"
#include "rclcpp/qos.hpp"
#include "std_msgs/msg/string.hpp"

using namespace std::chrono_literals;

/* This example creates a subclass of Node and uses std::bind() to register a
* member function as a callback from the timer. */
#define LEN_SET (200*1024)
class MinimalPublisher : public rclcpp::Node
{
  public:
    MinimalPublisher()
    : Node("minimal_publisher"), count_(0)
    {
      rclcpp::QoSInitialization qos_initialization = rclcpp::QoSInitialization::from_rmw(rmw_qos_profile_default);

      //publisher_ = this->create_publisher<std_msgs::msg::String>("topic", rclcpp::QoS(qos_initialization));
      publisher_ = this->create_publisher<std_msgs::msg::String>("topic", 10);
      timer_ = this->create_wall_timer(100ms, std::bind(&MinimalPublisher::timer_callback, this));
    }

  private:
    void timer_callback()
    {
      auto message = std_msgs::msg::String();
      std::string tmp(LEN_SET, 'a');
      message.data = tmp;
      //RCLCPP_INFO(this->get_logger(), "Publishing: '%s'", message.data.c_str());
      publisher_->publish(message);
		static int64_t old = 0;
  		auto tt = std::chrono::high_resolution_clock::now().time_since_epoch().count();
		std::cout << "time in nanosencond " << tt - old << std::endl;
	  old = tt;
    }
    rclcpp::TimerBase::SharedPtr timer_;
    rclcpp::Publisher<std_msgs::msg::String>::SharedPtr publisher_;
    size_t count_;
  };

  int main(int argc, char * argv[])
  {
    rclcpp::init(argc, argv);
    rclcpp::spin(std::make_shared<MinimalPublisher>());
    rclcpp::shutdown();
    return 0;
  }

after build ,use command "ros2 topic hz /topic" is gives out olny about 6 hz
and another example is velodyne_driver on nvidia xavier，link below:
https://github.com/ros-drivers/velodyne/issues/367
#### Expected behavior
ros2 topic hz /topic gives out almost as publish period ,eg publish at 100ms gives out about 10Hz 
#### Actual behavior

#### Additional information
https://github.com/ros-drivers/velodyne/issues/367

----
## Feature request

#### Feature description

#### Implementation considerations

fujitatomoya, 2020-10-30:
I was trying to reproduce this problem with source build https://github.com/ros2/ros2/commit/4aeefc197b1e7a4c267ae0e36147db15e565e849, but so far I cannot reproduce this issue in my environment.

```
root@ecbd40ccfda6:~/docker_ws/ros2_colcon# ros2 run ros2_test_prover rclcpp_1431 
...
[INFO] [1604059949.867247443] [minimal_publisher]: Publishing data...'
time in nano seconds 99920601
[INFO] [1604059949.967290480] [minimal_publisher]: Publishing data...'
time in nano seconds 100101309
[INFO] [1604059950.067301648] [minimal_publisher]: Publishing data...'
time in nano seconds 100015184
[INFO] [1604059950.167250440] [minimal_publisher]: Publishing data...'
time in nano seconds 99975538
[INFO] [1604059950.267200139] [minimal_publisher]: Publishing data...'
time in nano seconds 99818596

root@ecbd40ccfda6:~/docker_ws/ros2_colcon# ros2 topic hz /topic
...
average rate: 10.000
	min: 0.099s max: 0.101s std dev: 0.00022s window: 5120
average rate: 10.000
	min: 0.099s max: 0.101s std dev: 0.00022s window: 5130
average rate: 10.000
	min: 0.099s max: 0.101s std dev: 0.00022s window: 5141
average rate: 10.000
	min: 0.099s max: 0.101s std dev: 0.00022s window: 5152
average rate: 10.000
	min: 0.099s max: 0.101s std dev: 0.00022s window: 5162
```

average rate is 10Hz as expected.

849631323, 2020-11-02:
@fujitatomoya  thanks for your attention. 
I wonder to know that with which version you test ,is dashing or foxy ? in foxy it won't reproduce the bug until the payload more than 800KB. and I can't see your payload on you test result . 
in my environments , if  the payload eg, the publish  string size is 200KB or more than that will always reproduce the problem, what's more, the the payload of 16-line velodyne lidar is more than 500KB.

fujitatomoya, 2020-11-02:
@849631323 

> I wonder to know that with which version you test ,is dashing or foxy ?

master. this is said on my previous comment, `with source build ros2/ros2@4aeefc1`

> in foxy it won't reproduce the bug until the payload more than 800KB. and I can't see your payload on you test result .

i can check that with 800KB or bigger. btw, i just borrowed your test code and created reproducible project https://github.com/fujitatomoya/ros2_test_prover/commit/bfd07c872da2ed76b038cccf1746e6eb20ce51b4, so that we can easily just import and colcon build. as you can see, it is set with [200KB](https://github.com/fujitatomoya/ros2_test_prover/blob/bfd07c872da2ed76b038cccf1746e6eb20ce51b4/src/rclcpp_1431.cpp#L10) now.

fujitatomoya, 2020-11-02:
@849631323 

tested with 1MB data payload, I still cannot reproduce the problem.

> use command "ros2 topic hz /topic" is gives out olny about 6 hz

```
# ros2 run ros2_test_prover rclcpp_1431 
...<snip>
[INFO] [1604301534.067277740] [minimal_publisher]: Publishing data...'
time in nano seconds 100016364
[INFO] [1604301534.167274642] [minimal_publisher]: Publishing data...'
time in nano seconds 99996491
[INFO] [1604301534.267279186] [minimal_publisher]: Publishing data...'
time in nano seconds 100001474
[INFO] [1604301534.367284859] [minimal_publisher]: Publishing data...'
time in nano seconds 100013183
[INFO] [1604301534.467271852] [minimal_publisher]: Publishing data...'
time in nano seconds 99982113
[INFO] [1604301534.567117663] [minimal_publisher]: Publishing data...'
time in nano seconds 99841581

# ros2 topic hz /topic
...<snip>
average rate: 9.657
	min: 0.099s max: 0.600s std dev: 0.03131s window: 816
average rate: 9.661
	min: 0.099s max: 0.600s std dev: 0.03112s window: 826
average rate: 9.665
	min: 0.099s max: 0.600s std dev: 0.03092s window: 837
average rate: 9.669
	min: 0.099s max: 0.600s std dev: 0.03074s window: 847
average rate: 9.673
	min: 0.099s max: 0.600s std dev: 0.03054s window: 858
average rate: 9.677
	min: 0.099s max: 0.600s std dev: 0.03035s window: 869
average rate: 9.681
	min: 0.099s max: 0.600s std dev: 0.03018s window: 879
```

I see some jitter exists but I think that depends on the application how much accuracy it requires. maybe something you'd want to try is to use `rmw_cyclonedds` instead? you could refer to https://github.com/ros2/rmw_cyclonedds

849631323, 2020-11-05:
@fujitatomoya hi ,have you ever test on ARM platform such as nvidia TX2 or xavier, or raspberry pi
unfortunately I met the same issuse just as some one has test on xavier as below:
https://github.com/ros2/ros1_bridge/issues/218

fujitatomoya, 2020-11-05:
No, i just checked with a couple of Intel PC. i dont have arm64 platform for now.

y-okumura-isp, 2020-11-27:
I can reproduce with RPi4 8GB.

Environment
- master
- rclcpp: de353f9e45ca2ef191708e557f217eda843ab46a
- Fast-DDS: e7e96153a52d29e97c5a69a056bbd0cbb090b515

I test several cases:
- 2 process
  - 200KB payload -> 10 Hz
  - 1MB -> 9.6-10 Hz
  - 10MB -> 9.1-9.4 Hz (I describe resources below)
- 1 process
  - 200KB payload -> 10Hz (result is omitted in this post)
  - 10MB -> 9.6-10Hz

In 2 process setting, I cannot get any information by `ipcs` or `tcpdump`.
Is there any way to watch or get metrices of Fast-DDS communication?

----
200KB payload

```
# 200KB
$ ros2 topic hz /topic
average rate: 9.995
        min: 0.088s max: 0.111s std dev: 0.00550s window: 11
average rate: 10.003
        min: 0.088s max: 0.111s std dev: 0.00496s window: 22
average rate: 10.001
        min: 0.088s max: 0.111s std dev: 0.00467s window: 32
```

1 MB payload.

```
# 1MB, slightly get slow
$ ros2 topic hz /topic
average rate: 10.001
        min: 0.099s max: 0.101s std dev: 0.00057s window: 11
average rate: 10.003
        min: 0.098s max: 0.102s std dev: 0.00095s window: 22
average rate: 9.998
        min: 0.096s max: 0.107s std dev: 0.00178s window: 32
average rate: 9.999
        min: 0.096s max: 0.107s std dev: 0.00156s window: 43
average rate: 9.999
        min: 0.096s max: 0.107s std dev: 0.00146s window: 53
average rate: 9.999
        min: 0.096s max: 0.107s std dev: 0.00139s window: 64
average rate: 9.729
        min: 0.096s max: 0.201s std dev: 0.01656s window: 72
average rate: 9.646
        min: 0.096s max: 0.201s std dev: 0.01893s window: 82
```

10 MB payload.

```
$ ros2 topic hz /topic                                                                                                                                           
average rate: 9.100                                                                                                                                                                                                                  
        min: 0.094s max: 0.201s std dev: 0.03039s window: 10                                                                                                                                                                         
average rate: 9.025                                                                                                                                                                                                                  
        min: 0.094s max: 0.201s std dev: 0.03094s window: 19                                                                                                                                                                         
average rate: 9.073                                                                                                                                                                                                                  
        min: 0.091s max: 0.201s std dev: 0.03011s window: 29                                                                                                                                                                         
average rate: 9.287                                                                                                                                                                                                                  
        min: 0.091s max: 0.201s std dev: 0.02636s window: 39                                                                                                                                                                         
average rate: 9.440                                                                                                                                                                                                                  
        min: 0.091s max: 0.201s std dev: 0.02352s window: 50
[... 31 lines cut ...]
%Cpu1  : 18.9 us, 22.6 sy,  0.0 ni, 58.5 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
%Cpu2  : 26.2 us,  9.6 sy,  0.0 ni, 64.1 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
%Cpu3  : 28.1 us, 18.2 sy,  0.0 ni, 53.6 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st

$ vmstat 1
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 2  0      0 1970628 340160 5146244    0    0     0     6    2    1  1  0 98  0  0
 1  0      0 1969620 340160 5146244    0    0     0     0 3249 5336 22 11 67  0  0
 0  0      0 1968612 340160 5146244    0    0     0     0 4251 7341 22 15 63  0  0
 0  0      0 1967604 340160 5146244    0    0     0     0 5377 9514 27 18 55  0  0
 2  0      0 1944672 340160 5146244    0    0     0     0 3697 6185 20 12 67  0  0
 0  0      0 1967636 340160 5146244    0    0     0    28 5368 9437 28 16 56  0  0
 0  0      0 1967384 340160 5146244    0    0     0     0 3489 5632 24 15 62  0  0
 
$ sudo ipcs
------ Message Queues --------
key        msqid      owner      perms      used-bytes   messages    

------ Shared Memory Segments --------
key        shmid      owner      perms      bytes      nattch     status      

------ Semaphore Arrays --------
key        semid      owner      perms      nsems   
```

y-okumura-isp, 2020-11-30:
I got extra results.

## Summary

(1) Only 5-6 Hz is obtained in CycloneDDS 10 MB x 10 Hz
(2) 700 Mbps is a limit of RPi4, loopback UDP communication. So, 10 MB x 10 Hz looks a too high goal to RPi4 in my setting.
(3) About 9.7 Hz is obtained in 8 MB x 10 Hz which is lower then 700 Mbps.
(4) But **changing kernel parameter, CyclonDDS can do 8 MB x 10 Hz**. FastDDS cannot.

Here are the details.

## (1) CycloneDDS @ 10 MB x 10 Hz => 5-6 hz
Here is the result of CycloneDDS @ 10 MB x 10Hz. 5-6 Hz is too low compared to 9.1-9.5 Hz of FastDDS.
What's the difference?

```
# term1, argv[1] means message size
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_pub 10000000

# term2
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_sub
  
$ ros2 topic hz /topic
average rate: 5.763
        min: 0.106s max: 0.214s std dev: 0.03865s window: 7
average rate: 5.675
        min: 0.106s max: 0.217s std dev: 0.03624s window: 13
average rate: 5.032
        min: 0.106s max: 0.351s std dev: 0.06440s window: 18
average rate: 4.944
        min: 0.106s max: 0.449s std dev: 0.08084s window: 23
```

## (2) 700 Mbps is a limit of RPi4
10 MB x 10 Hz is 800 Mbps. So I checked by `iperf` then packet drop happens.
Additionally, it seems 706 Mbps is possible.
706 Mbps was the same after the tuning of (4). I think we need to tune other parameters such as jumbo frames.

```
$ iperf -u -c localhost -b 800m -i 1
$ iperf -s -u -i 1
------------------------------------------------------------
Server listening on UDP port 5001
Receiving 1470 byte datagrams
UDP buffer size:  208 KByte (default)
------------------------------------------------------------
[ ID] Interval       Transfer     Bandwidth        Jitter   Lost/Total Datagrams
[  3]  0.0- 1.0 sec  84.2 MBytes   706 Mbits/sec   0.001 ms 8026/68065 (12%)
[  3]  1.0- 2.0 sec  84.3 MBytes   707 Mbits/sec   0.001 ms 7878/68029 (12%)
[  3]  2.0- 3.0 sec  84.4 MBytes   708 Mbits/sec   0.001 ms 7820/68025 (11%)
[  3]  3.0- 4.0 sec  84.4 MBytes   708 Mbits/sec   0.000 ms 7843/68028 (12%)
```

## (3) 8 MB x 10 Hz experiment
I tried 8 MB x 10 Hz because 8 MB x 10 Hz = 80 MB/s = 640 Mbps < 706 Mbps. 
But it looks too high by CycloneDDS and FastDDS (maybe because of PTPS overhead).
Both have 9.6 - 9.8 Hz.

This is FastDDS results. At first, it has 10 Hz but gradually slows down.

```
./build/rclcpp_1431/main_pub 8000000
./build/rclcpp_1431/main_sub

average rate: 10.054
        min: 0.090s max: 0.108s std dev: 0.00593s window: 11
average rate: 10.026
        min: 0.090s max: 0.108s std dev: 0.00502s window: 21
(snip)
average rate: 9.771
        min: 0.090s max: 0.199s std dev: 0.01535s window: 83
average rate: 9.793
        min: 0.090s max: 0.199s std dev: 0.01454s window: 93
(snip)
average rate: 9.764
        min: 0.090s max: 0.207s std dev: 0.01568s window: 165
average rate: 9.725
        min: 0.090s max: 0.207s std dev: 0.01690s window: 175
```

## (4) Changing kernel parameter enables 8 MB x 10 Hz with CycloneDDS

Linux kernel has some parameters about UDP. According to this article, I changed `rmem_max` and `rmem_default`.
https://medium.com/@CameronSparr/increase-os-udp-buffers-to-improve-performance-51d167bb1360

```
# set to 26 MB
sudo sysctl -w net.core.rmem_max=26214000
sudo sysctl -w net.core.rmem_default=26214000
```

Then CycloneDDS can communicate at 8 MB x 10 Hz.

```
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
./build/rclcpp_1431/main_pub 8000000

# term2
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_sub

$ ros2 topic hz /topic
average rate: 10.003
        min: 0.088s max: 0.107s std dev: 0.00456s window: 11
average rate: 9.997
        min: 0.088s max: 0.107s std dev: 0.00337s window: 21
average rate: 10.001
        min: 0.088s max: 0.107s std dev: 0.00278s window: 32
(snip)
average rate: 10.000
        min: 0.088s max: 0.107s std dev: 0.00173s window: 104
average rate: 9.999
        min: 0.088s max: 0.107s std dev: 0.00168s window: 114
```

The result of FastDDS is unchanged (lower than 10 Hz)

```
./build/rclcpp_1431/main_pub 8000000
./build/rclcpp_1431/main_sub

average rate: 10.092
        min: 0.084s max: 0.111s std dev: 0.00622s window: 11
average rate: 9.565
        min: 0.084s max: 0.202s std dev: 0.02284s window: 20
average rate: 9.703
        min: 0.084s max: 0.202s std dev: 0.01893s window: 30
average rate: 9.781
        min: 0.084s max: 0.202s std dev: 0.01626s window: 41
```

ivanpauno, 2021-01-08:
@y-okumura-isp based on the comments this seems to be a rmw vendor dependent "network tweaking" problem, is that right?
IIUC the UDP max buffer size on linux is "small", and under large data size or medium data size and high rate you need to either tweak the kernel buffer sizes or configure your DDS implementation in a way that "doesn't send data that fast".

(in my experience Fast-RTPS defaults seem to work better with large data which is consistent with your results above)

Though I think that this is important for the overall "ROS 2 experience" (considering that sending large data is quite usual in robotics ...), I don't think we can do much about it in `rclcpp`. In that case, I would close this issue.

mabelzhang, 2021-03-09:
Closing based on comment above. Feel free to reopen.