ros2/rclcpp #1455: Segmentation fault on wait_for_work

v-lopez, 2020-11-13:
## Bug report

As a workaround for https://github.com/ros2/rclcpp/issues/1454 I tried to create a utility to avoid a hard coded sleep, that blocks until the spin() is active and a callback has been called, before returning.
It causes a segmentation fault very unfrequently.

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries
- Version or commit hash: 5.0.0-1focal.20201007.211227
- DDS implementation: Fast DDS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
``` 
std::future<void> spin_executor_async(std::shared_ptr<rclcpp::Node> node, rclcpp::Executor &executor)
{
  // We create a timer that will be called as part of the spin
  // this function won't return until the executor is spinning and the timer is called
  std::atomic<bool> timer_called{false};
  std::function<void()> timer_called_lambda = [&](){std::cout << " timer called " << std::endl; timer_called = true;};
  auto timer = node->create_wall_timer(std::chrono::nanoseconds(1), timer_called_lambda);
  executor.add_node(node);
  auto executor_spin_future = std::async(
    std::launch::async, [&executor]() -> void {
      executor.spin();
    });

  while (!timer_called)
  {
    std::this_thread::sleep_for(1ns);
  }
  timer.reset();
  executor.remove_node(node);
  return executor_spin_future;
}

TEST(Foo, bar2) {
  auto node = std::make_shared<rclcpp::Node>("test");
  auto executor = std::make_shared<rclcpp::executors::SingleThreadedExecutor>();

  auto executor_spin_future = spin_executor_async(node, *executor);
  executor->cancel();
  return;
}
```

#### Expected behavior
That the test doesn't hang, as a workaround for #1454 

#### Actual behavior
Doesn't hang, but very unfrequently 1% of runs it segfaults.
Because the thread is stuck on `executor->spin()` (wait_for_work exactly)

#### Additional information
The segfault happens on thread 2 (the executor spin thread), here's the callstack:
```
Thread 2 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 9204.9221]
0x00007fc84fcd007b in rclcpp::memory_strategies::allocator_memory_strategy::AllocatorMemoryStrategy<std::allocator<void> >::collect_entities(std::map<std::weak_ptr<rclcpp::CallbackGroup>, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::owner_less<std::weak_ptr<rclcpp::CallbackGroup> >, std::allocator<std::pair<std::weak_ptr<rclcpp::CallbackGroup> const, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface> > > > const&) () from /opt/ros/rolling/lib/librclcpp.so
(rr) bt
#0  0x00007fc84fcd007b in rclcpp::memory_strategies::allocator_memory_strategy::AllocatorMemoryStrategy<std::allocator<void> >::collect_entities(std::map<std::weak_ptr<rclcpp::CallbackGroup>, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface>, std::owner_less<std::weak_ptr<rclcpp::CallbackGroup> >, std::allocator<std::pair<std::weak_ptr<rclcpp::CallbackGroup> const, std::weak_ptr<rclcpp::node_interfaces::NodeBaseInterface> > > > const&) () from /opt/ros/rolling/lib/librclcpp.so
#1  0x00007fc84fcb93b6 in rclcpp::Executor::wait_for_work(std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /opt/ros/rolling/lib/librclcpp.so
#2  0x00007fc84fcb9da3 in rclcpp::Executor::get_next_executable(rclcpp::AnyExecutable&, std::chrono::duration<long, std::ratio<1l, 1000000000l> >) () from /opt/ros/rolling/lib/librclcpp.so
#3  0x00007fc84fcbf00d in rclcpp::executors::SingleThreadedExecutor::spin() () from /opt/ros/rolling/lib/librclcpp.so
#4  0x0000563c231efd7d in <lambda()>::operator()(void) const (__closure=0x563c23c088e8) at /root/control/src/ros2_control/controller_manager/test/test_controller_manager.cpp:143
#5  0x0000563c231f2e13 in std::__invoke_impl<void, spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> >(std::__invoke_other, <lambda()> &&) (__f=...)
    at /usr/include/c++/9/bits/invoke.h:60
#6  0x0000563c231f2db4 in std::__invoke<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> >(<lambda()> &&) (__fn=...) at /usr/include/c++/9/bits/invoke.h:95
#7  0x0000563c231f2d44 in std::thread::_Invoker<std::tuple<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> > >::_M_invoke<0>(std::_Index_tuple<0>) (this=0x563c23c088e8)
[... 39 lines cut ...]
   from /opt/ros/rolling/lib/librclcpp.so
#3  0x00007fc84fcb8f4d in rclcpp::Executor::remove_node(std::shared_ptr<rclcpp::node_interfaces::NodeBaseInterface>, bool) () from /opt/ros/rolling/lib/librclcpp.so
#4  0x00007fc84fcb42e4 in rclcpp::Executor::remove_node(std::shared_ptr<rclcpp::Node>, bool) () from /opt/ros/rolling/lib/librclcpp.so
#5  0x0000563c231eff4c in spin_executor_async (node=std::shared_ptr<class rclcpp::Node> (use count 3, weak count 1) = {...}, executor=...)
    at /root/control/src/ros2_control/controller_manager/test/test_controller_manager.cpp:151
#6  0x0000563c231f00ad in TestControllerManager_controller_parameters_Test::TestBody (this=0x563c23d26bd0) at /root/control/src/ros2_control/controller_manager/test/test_controller_manager.cpp:177
#7  0x0000563c23241c34 in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void> (object=0x563c23d26bd0, method=&virtual testing::Test::TestBody(), 
    location=0x563c2326d73b "the test body") at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2447
#8  0x0000563c2323a9f9 in testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void> (object=0x563c23d26bd0, method=&virtual testing::Test::TestBody(), location=0x563c2326d73b "the test body")
    at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2483
#9  0x0000563c23216646 in testing::Test::Run (this=0x563c23d26bd0) at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2522
#10 0x0000563c2321703f in testing::TestInfo::Run (this=0x563c238d4f30) at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2703
#11 0x0000563c23217730 in testing::TestCase::Run (this=0x563c238d4db0) at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2825
#12 0x0000563c23222e19 in testing::internal::UnitTestImpl::RunAllTests (this=0x563c238d4a40) at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:5216
#13 0x0000563c2324312e in testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x563c238d4a40, 
    method=(bool (testing::internal::UnitTestImpl::*)(class testing::internal::UnitTestImpl * const)) 0x563c23222b70 <testing::internal::UnitTestImpl::RunAllTests()>, 
    location=0x563c2326e140 "auxiliary test code (environments or event listeners)") at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2447
#14 0x0000563c2323bb0d in testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool> (object=0x563c238d4a40, 
    method=(bool (testing::internal::UnitTestImpl::*)(class testing::internal::UnitTestImpl * const)) 0x563c23222b70 <testing::internal::UnitTestImpl::RunAllTests()>, 
    location=0x563c2326e140 "auxiliary test code (environments or event listeners)") at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:2483
#15 0x0000563c232217f8 in testing::UnitTest::Run (this=0x563c232b3520 <testing::UnitTest::GetInstance()::instance>) at /opt/ros/rolling/src/gtest_vendor/src/gtest.cc:4824
#16 0x0000563c2320e295 in RUN_ALL_TESTS () at /opt/ros/rolling/src/gtest_vendor/include/gtest/gtest.h:2370
#17 0x0000563c2320e217 in main (argc=1, argv=0x7fff2d24dfa8) at /opt/ros/rolling/src/gmock_vendor/src/gmock_main.cc:53

```

ivanpauno, 2020-11-13:
My guess is that removing the timer while spinning (`timer.reset();`) might not be thread safe, but I'm not sure if that's the issue.

@v-lopez could you try building with debug info to get more information in the backtrace? (i.e., line numbers), thanks!

v-lopez, 2020-11-16:
Adding debug callstack from rclcpp master (dd0f97f1790de0384caeeb94a7c1c49e4b9f08ed):

Thread 2:
```
Thread 2 received signal SIGSEGV, Segmentation fault.
[Switching to Thread 18773.18784]
0x00007f15ee02fb12 in rclcpp::Executor::add_callback_groups_from_nodes_associated_to_executor (this=0x55cc2bc42600) at /root/rclcpp/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:175
175	      auto group_ptrs = node->get_callback_groups();
(rr) bt
#0  0x00007f15ee02fb12 in rclcpp::Executor::add_callback_groups_from_nodes_associated_to_executor (this=0x55cc2bc42600) at /root/rclcpp/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:175
#1  0x00007f15ee033bae in rclcpp::Executor::wait_for_work (this=0x55cc2bc42600, timeout=...) at /root/rclcpp/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:664
#2  0x00007f15ee034d52 in rclcpp::Executor::get_next_executable (this=0x55cc2bc42600, any_executable=..., timeout=...) at /root/rclcpp/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:871
#3  0x00007f15ee045459 in rclcpp::executors::SingleThreadedExecutor::spin (this=0x55cc2bc42600) at /root/rclcpp/src/ros2/rclcpp/rclcpp/src/rclcpp/executors/single_threaded_executor.cpp:35
#4  0x000055cc2b5e9f6b in <lambda()>::operator()(void) const (__closure=0x55cc2bc424f8) at /root/rclcpp/src/ros2/rclcpp/rclcpp/test/rclcpp/test_duration.cpp:278
#5  0x000055cc2b5eca27 in std::__invoke_impl<void, spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> >(std::__invoke_other, <lambda()> &&) (__f=...) at /usr/include/c++/9/bits/invoke.h:60
#6  0x000055cc2b5ec9c8 in std::__invoke<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> >(<lambda()> &&) (__fn=...) at /usr/include/c++/9/bits/invoke.h:95
#7  0x000055cc2b5ec958 in std::thread::_Invoker<std::tuple<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> > >::_M_invoke<0>(std::_Index_tuple<0>) (this=0x55cc2bc424f8) at /usr/include/c++/9/thread:244
#8  0x000055cc2b5ec8e5 in std::thread::_Invoker<std::tuple<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> > >::operator()(void) (this=0x55cc2bc424f8) at /usr/include/c++/9/thread:251
#9  0x000055cc2b5ec688 in std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> > >, void>::operator()(void) const (this=0x7f15e4c4b5d0) at /usr/include/c++/9/future:1362
#10 0x000055cc2b5ec38b in std::_Function_handler<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>(), std::__future_base::_Task_setter<std::unique_ptr<std::__future_base::_Result<void>, std::__future_base::_Result_base::_Deleter>, std::thread::_Invoker<std::tuple<spin_executor_async(std::shared_ptr<rclcpp::Node>, rclcpp::Executor&)::<lambda()> > >, void> >::_M_invoke(const std::_Any_data &) (__functor=...) at /usr/include/c++/9/bits/std_function.h:286
#11 0x000055cc2b5f06d4 in std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>::operator()() const (this=0x7f15e4c4b5d0) at /usr/include/c++/9/bits/std_function.h:688
#12 0x000055cc2b5ef0be in std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*) (this=0x55cc2bc424c0, __f=0x7f15e4c4b5d0, __did_set=0x7f15e4c4b52f)
    at /usr/include/c++/9/future:561
#13 0x000055cc2b5f569d in std::__invoke_impl<void, void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(std::__invoke_memfun_deref, void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__f=
    @0x7f15e4c4b570: (void (std::__future_base::_State_baseV2::*)(std::__future_base::_State_baseV2 * const, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter>()> *, bool *)) 0x55cc2b5ef084 <std::__future_base::_State_baseV2::_M_do_set(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*)>, __t=@0x7f15e4c4b540: 0x55cc2bc424c0) at /usr/include/c++/9/bits/invoke.h:73
#14 0x000055cc2b5f32ae in std::__invoke<void (std::__future_base::_State_baseV2::*)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*>(void (std::__future_base::_State_baseV2::*&&)(std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*, bool*), std::__future_base::_State_baseV2*&&, std::function<std::unique_ptr<std::__future_base::_Result_base, std::__future_base::_Result_base::_Deleter> ()>*&&, bool*&&) (__fn=
[... cut ...]
    at /root/rclcpp/install/gtest_vendor/src/gtest_vendor/./src/gtest.cc:2469
#19 0x000055cc2b611f64 in testing::UnitTest::Run (this=0x55cc2b67c560 <testing::UnitTest::GetInstance()::instance>) at /root/rclcpp/install/gtest_vendor/src/gtest_vendor/./src/gtest.cc:4925
#20 0x000055cc2b5fdfa4 in RUN_ALL_TESTS () at /root/rclcpp/install/gtest_vendor/src/gtest_vendor/include/gtest/gtest.h:2473
#21 0x000055cc2b5fdf26 in main (argc=1, argv=0x7ffdbf4ea218) at /root/rclcpp/install/gtest_vendor/src/gtest_vendor/src/gtest_main.cc:45
```

ivanpauno, 2020-11-16:
Thanks for the additional information @v-lopez !

This seems to have been introduced in https://github.com/ros2/rclcpp/pull/1218.
It might be the same issue than in https://github.com/ros2/rclcpp/issues/1402, or a different one.

@wjwwood FYI

wjwwood, 2020-11-17:
I'll have a look ASAP. Thanks @ivanpauno

fujitatomoya, 2020-11-18:
according to the stack trace, I think this is because [weak_nodes_](https://github.com/ros2/rclcpp/blob/de353f9e45ca2ef191708e557f217eda843ab46a/rclcpp/include/rclcpp/executor.hpp#L566-L567) is not protected by mutex.

brawner, 2020-12-02:
#1402 has been resolved. But I think this issue is different. Does the issue still occur with the most recent commits?

v-lopez, 2020-12-02:
Yes it does still crash as of ea0ee50318a9609843cd7295c4542c42a7ed339c

ivanpauno, 2021-01-08:
@v-lopez I would expect https://github.com/ros2/rclcpp/pull/1505 to fix the issue, could you give it a try?

v-lopez, 2021-01-08:
Before I would get a crash in a few seconds of attempts, now it took over a minute but I still got a segfault. 

Running it with valgrind to see if I can catch it.

ivanpauno, 2021-01-08:
@v-lopez I tried your example locally and I think that after last #1505 commit I got it right.
(I'm still going to run the reproducer for some minutes more, as in my system it was hard to make the example segfault)

v-lopez, 2021-01-08:
Without these latest changes, it took me 4h of gdb to catch a crash, but I'm attaching the callstack.

[callstack.txt](https://github.com/ros2/rclcpp/files/5789666/callstack.txt)

I'm running now with the latest changes and so far no crashes. I'd consider it fixed.

ivanpauno, 2021-01-11:
> I'm running now with the latest changes and so far no crashes. I'd consider it fixed.

I'm confident that it's fixed now as that callstack is the same I got before the latest commit (segmentation fault while copy constructing), so we only need to wait for a review of #1505 :tada: .

ivanpauno, 2021-01-19:
> @ivanpauno With #1505 now merged, can we close this?

Yes, closing

closed 2021-01-19 by ivanpauno