ros2/rclcpp #1460: [rclcpp_lifecycle] The get_available_transitions method and get_available_transitions service produce different results

brawner, 2020-11-16:
I noticed this while writing up the benchmarks, but calling `rclcpp_lifecycle::LifecycleNode::get_available_transitions()` will return all 25 transitions, but calling the `get_available_transitions` service will only return the available transitions for the current state. In order to get all 25 transitions from a service, you need to call the `get_transition_graph` service.

Perhaps it would make sense to add a `get_transition_graph` method to the lifecycle node and update `get_available_transitions` to only return the available transitions of the current state so that it matches the similarly named services?

Thoughts maintainers/author? @wjwwood @ivanpauno @mabelzhang @Karsten1987

wjwwood, 2020-11-17:
Makes sense to me to make it consistent between the two.

fujitatomoya, 2020-11-22:
@brawner @wjwwood 

could you review https://github.com/ros2/rclcpp/pull/1472 when you got time?

fujitatomoya, 2020-12-02:
@brawner 

i think that we can close this.

brawner, 2020-12-02:
Resolved by #1472, thanks @fujitatomoya

closed 2020-12-02 by brawner