ros2/rclcpp #1478: "UnawareGoalHandleError" exception thrown when action client tries to wait for goal result future

schornakj, 2020-11-25:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04 LTS
- Installation type:
  - Binaries
- Version or commit hash:
  - Eloquent
- DDS implementation:
  - CycloneDDS, FastRTPS
- Client library (if applicable):
  - rclcpp

#### Summary

I have an action client/server pair that handles a large number of goals in parallel. The client node eventually throws an `'rclcpp_action::exceptions::UnawareGoalHandleError` exception even though I have set callbacks for both the goal response and the goal result.

#### Steps to reproduce issue

I created action client and server nodes to demonstrate this behavior in the `feature/parallel-client` branch of this repo: https://github.com/schornakj/rclcpp_action_exception_test/tree/feature/parallel-client (the same repo from #1399).

Here's the source code from my nodes:

**`client_node_parallel.h`**

```
#pragma once

#include <rclcpp/rclcpp.hpp>
#include <rclcpp_action/rclcpp_action.hpp>
#include <example_interfaces/action/fibonacci.hpp>

class ClientNodeParallel : public rclcpp::Node
{
public:
  using Fibonacci = example_interfaces::action::Fibonacci;
  using ClientGoalHandleFibonacci = rclcpp_action::ClientGoalHandle<Fibonacci>;

  explicit ClientNodeParallel();

private:
  void sendGoal();

  void asyncHandleClientReq(const std::size_t& order, const rclcpp_action::Client<Fibonacci>::SharedPtr& client);

  rclcpp::callback_group::CallbackGroup::SharedPtr cb_group_mutually_exclusive_;
  rclcpp_action::Client<Fibonacci>::SharedPtr client_;
  rclcpp::TimerBase::SharedPtr timer_;
  std::atomic_uint n_reqs_in_progress_;
  const std::size_t max_parallel_reqs_;
  std::atomic_uint n_completed_;
};
```

**`client_node_parallel.cpp`**

```
#include <action_client_test/client_node_parallel.h>

using std::placeholders::_1;
using std::placeholders::_2;

ClientNodeParallel::ClientNodeParallel()
  : Node("action_test_client_node")
  , cb_group_mutually_exclusive_(create_callback_group(rclcpp::callback_group::CallbackGroupType::MutuallyExclusive))
  , client_(rclcpp_action::create_client<Fibonacci>(this, "/fibonacci", cb_group_mutually_exclusive_))
  , timer_(create_wall_timer(std::chrono::duration<double>(0.01), std::bind(&ClientNodeParallel::sendGoal, this)))
  , n_reqs_in_progress_{0}
[... 180 lines cut ...]
ros2 launch action_exception_test run_nodes_parallel.launch.py 
[INFO] [launch]: All log files can be found below /home/joe/.ros/log/2020-11-25-14-01-21-762074-yakshaver-19006
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [action_exception_test_server_node-1]: process started with pid [19028]
[INFO] [action_exception_test_client_node_parallel-2]: process started with pid [19029]
[action_exception_test_client_node_parallel-2] [INFO] [client_node]: N_IN_PROGRESS=0
[action_exception_test_client_node_parallel-2] [INFO] [client_node]: Action #1 succeeded! 0 1 1 2 3 5 8 13 21 34 55 89 144 233 377 610 

(many successful actions later...)

[action_exception_test_client_node_parallel-2] [INFO] [client_node]: Action #1121 succeeded! 0 1 1 2 3 5 8 13 21 34 55 89 144 233 377 610 
[action_exception_test_client_node_parallel-2] [INFO] [client_node]: N_IN_PROGRESS=39

[action_exception_test_client_node_parallel-2] terminate called after throwing an instance of 'rclcpp_action::exceptions::UnawareGoalHandleError'
[action_exception_test_client_node_parallel-2]   what():  Goal handle is not tracking the goal result.
[ERROR] [action_exception_test_client_node_parallel-2]: process has died [pid 27920, exit code -6, cmd '/home/joe/workspaces/cb_group_test_ws/install/action_exception_test/lib/action_exception_test/action_exception_test_client_node_parallel --ros-args -r __node:=client_node'].
```

The error message in the exception suggests that it is related to my use of the async goal result future in the action client. I've gotten the same exception in cases where I was waiting on that future without having set a goal result callback, but in this case I  have set a goal result callback (albeit using a trivial empty function).

#### Additional information

I have also experienced this error in systems with more traditional client/server relationships, where the client sends just one goal request at a time. In that case the time between exceptions is much longer (weeks!), which I think suggests that this error has some constant small probability of being thrown for each action goal request that I send.

The nodes will need to be modified to be compatible with the current `rclcpp` master branch (if I have time this weekend I'll do this myself in a new branch).

fujitatomoya, 2020-11-29:
@schornakj 

> The nodes will need to be modified to be compatible with the current rclcpp master branch (if I have time this weekend I'll do this myself in a new branch).

done, we can run parallel test based on https://github.com/fujitatomoya/rclcpp_action_exception_test/tree/feature/parallel-client-master. (create PR if you want to take it, https://github.com/schornakj/rclcpp_action_exception_test/pull/2)

so far, i cannot reproduce the problem... could you share core stack trace if you can?

schornakj, 2020-12-01:
> so far, i cannot reproduce the problem... could you share core stack trace if you can?

Interesting! I haven't tried this on the rclcpp master yet. If this issue only shows up in Eloquent does that imply that the root cause might have been fixed elsewhere? I see that there's been a lot of work targeting threading issues in the rclcpp action handling code.

I'll get a stack trace for you. Do I need to build ROS2 from source to get sufficiently useful information?

fujitatomoya, 2020-12-01:
> I'll get a stack trace for you.

thanks 👍 

> Do I need to build ROS2 from source to get sufficiently useful information?

I think if we could reproduce this issue based on master:HEAD, that would be really useful. but as you mentioned on previous comment, this might be fixed in master. there is a possibility this depends on environment, so it would be really nice to try with source 😄

clalancette, 2020-12-17:
@schornakj Friendly ping, have you been able to try this out on the latest?  You could either build from source or use the rolling binaries as a test.

schornakj, 2020-12-17:
@clalancette I've been tied up with other work for the past couple weeks, sorry about that.

clalancette, 2020-12-17:
> @clalancette I've been tied up with other work for the past couple weeks, sorry about that.

No worries.  Feedback when you have a chance is appreciated.

clalancette, 2022-02-03:
No response on this for over a year, so closing this out.  Feel free to comment or reopen if you are still having a problem.

closed 2022-02-03 by clalancette