ros2/rclcpp #1482: inconsistent parameter service behavior with non-exist parameter request.

fujitatomoya, 2020-11-27:
## Bug report

- Operating System:
  - ubuntu20.04
- Installation type:
  - source build
- Version or commit hash:
  - https://github.com/ros2/ros2/commit/c059de9aadbf3d9ad189ca17d62f0110683c0a65
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

once node is created, there are built-in services to access parameters. there seems to be inconsistent behavior for these services between `describe_parameters` / `get_parameter_types` and `get_parameters`. `describe_parameters` / `get_parameter_types` services return empty if request include non-exist parameter name. on the other hand `get_parameters` always returns the same size with request which includes non-exist parameter name.

#### Steps to reproduce issue

- describe_parameters

```
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/describe_parameters rcl_interfaces/srv/DescribeParameters "names: [use_sim_time, not_exist]"
waiting for service to become available...
requester: making request: rcl_interfaces.srv.DescribeParameters_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.DescribeParameters_Response(descriptors=[])
```

-> empty since there is undeclared parameter `non_exist`. (it prints WARN message on server that failed to describe `non_exist`)

- get_parameter_types

```
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/get_parameter_types rcl_interfaces/srv/GetParameterTypes "names: [use_sim_time, not_exist]"
waiting for service to become available...
requester: making request: rcl_interfaces.srv.GetParameterTypes_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.GetParameterTypes_Response(types=[])
```

-> empty since there is undeclared parameter `non_exist`.  (it prints WARN message on server that failed to get type `non_exist`)

- get_parameters

```
root@5c7a784cba7f:~/ros2_ws/colcon_ws# ros2 service call /talker/get_parameters rcl_interfaces/srv/GetParameters "names: [use_sim_time, not_exist]"
requester: making request: rcl_interfaces.srv.GetParameters_Request(names=['use_sim_time', 'not_exist'])

response:
rcl_interfaces.srv.GetParameters_Response(values=[rcl_interfaces.msg.ParameterValue(type=1, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[]), rcl_interfaces.msg.ParameterValue(type=0, bool_value=False, integer_value=0, double_value=0.0, string_value='', byte_array_value=[], bool_array_value=[], integer_array_value=[], double_array_value=[], string_array_value=[])])
```

-> return default Parameter for `non_exist` request.

#### Expected behavior

I think it would be nice to have consistent behavior based on `NodeParameters::allow_undeclared_`(default false)

#### Actual behavior

behavior is inconsistent.

#### Additional information

further more, there is not even consistent with rclpy. if we use rclpy and provided built-in parameter services,

- `get_parameters` returns the same size with request, it will return default `ParameterValue` for `non_exist`. (same with rclcpp)
- **`describe_parameters` returns the same size with request, it will return default `ParameterDescriptor` for for `non_exist`.**
- **`get_parameter_types` returns the same size with request, it will return default `type = 0` for for `non_exist`.**

fujitatomoya, 2020-11-27:
CC: @clalancette

brawner, 2020-12-15:
@fujitatomoya can this be closed?

fujitatomoya, 2020-12-15:
@brawner 
CC: @clalancette 

no, not yet. I didn't start implementation, still waiting for design comment. (see https://github.com/ros2/rclcpp/pull/1453#issuecomment-734687708 and https://github.com/ros2/rclcpp/pull/1453#issuecomment-743359918)

ivanpauno, 2021-01-08:
Closing as linked PRs were merged, feel free to reopen if this is not solved.

closed 2021-01-08 by ivanpauno