ros2/rclcpp #1483: expose rcl_graph interfaces with test cases

fujitatomoya, 2020-11-30:
## Feature request

#### Feature description

The following [rcl_graph](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/graph.h) interfaces are not exposed to `rclcpp::node_interfaces::NodeGraph`.

- rcl_get_publisher_names_and_types_by_node
- rcl_get_subscriber_names_and_types_by_node
- rcl_get_client_names_and_types_by_node
- rcl_get_node_names_with_enclaves

#### Implementation considerations

Implementation should be straight-forward to just bind the interfaces from `rcl` to `rclcpp`.

#### Note

`rclpy` does have all of the [rcl_graph](https://github.com/ros2/rcl/blob/master/rcl/include/rcl/graph.h).

fujitatomoya, 2020-12-01:
@ivanpauno 

not urgent. but when you got time, could you take a look and share your thougths?

ivanpauno, 2020-12-18:
I'm marking this as `help wanted` `good first issue`, as anybody that need those methods exposed in `rclcpp` can create a PR.

Up to now, those were only exposed in `rclpy` and are mainly being used in the `ros2cli` tools (I don't know of much user code making use of them directly).

ivanpauno, 2020-12-18:
Ah, I didn't see there was a PR already open :grimacing:

closed 2021-09-23 by fujitatomoya