ros2/rclcpp #1485: publishing is very slow in 1 publisher and 2 subscribers situation at RPi4

y-okumura-isp, 2020-12-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04, Raspberry Pi 4
- Installation type:
  - build
- Version or commit hash:
  - de353f9e45ca2ef191708e557f217eda843ab46a
- DDS implementation:
  - CycloneDDS 9842f92fea76e8cc9a03a6fa40472b0ac0cca682
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I am investigating https://github.com/ros2/rclcpp/issues/1431, and found publishing is slow in one publisher and two subscribers situation. I could reproduce this on RPi4 and CycloneDDS.
We couldn't reproduce this issue on PC (i7-6800K).
Additionally, we couldn't reproduce too with RPi4 + FastDDS.
(I report to this repository so far because I use rclcpp)

Build source as the following.

```
git clone https://github.com/y-okumura-isp/ROS2_Samples
cd ROS2_Samples/rclcpp_1431
colcon build --symlink-install
```

Run one publisher and two subscribers.
In the commands below, data size is 200KB with 10 Hz.
When I run this with one publisher and one subscriber, I got 10Hz.

```
# term 1
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_sub

# term 2
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_sub

# term 3
RMW_IMPLEMENTATION=rmw_cyclonedds_cpp \
  ./build/rclcpp_1431/main_pub 200000

# term 4
ros2 topic hz /topic
average rate: 1.354
        min: 0.702s max: 0.805s std dev: 0.04692s window: 3
average rate: 1.251
        min: 0.607s max: 1.175s std dev: 0.19800s window: 5
average rate: 1.197
        min: 0.607s max: 1.175s std dev: 0.17694s window: 7
```

#### Expected behavior
We can get 10 Hz.

#### Actual behavior
About 1 Hz.

#### Additional information
CPU/Memory are not so used.

```
$ vmstat 1
procs -----------memory---------- ---swap-- -----io---- -system-- ------cpu-----
 r  b   swpd   free   buff  cache   si   so    bi    bo   in   cs us sy id wa st
 1  0      0 1608296 360264 5761196    0    0     0     5    2    5  2  0 97  0  0
 0  0      0 1608044 360264 5761196    0    0     0     0  324  404  1  1 99  0  0
 0  0      0 1607600 360264 5761196    0    0     0    24  320  344  0  1 99  0  0
 0  0      0 1607856 360264 5761196    0    0     0     0  331  414  1  1 99  0  0
 0  0      0 1607256 360264 5761196    0    0     0     0  386  470  1  0 99  0  0
```

I changed QoS BestEffort(1), but the situation was the same.
When I used two RPi4 (one for publisher, one for subscribers), the situation was the same too.

y-okumura-isp, 2020-12-15:
Hmm, after I rebooted my RPi4,I can not reproduce this issue.
So I close this. Sorry for the confusion.

closed 2020-12-15 by y-okumura-isp

cfzhous, 2022-05-11:
I had the same problem on Jetson AGX Xavier platform, don't know how you solved it？

y-okumura-isp, 2022-05-11:
@cfzhous Sorry, I don't know the reason. It disappeared after reboot.

Are there other nodes in your network?
The output of `ros2 node list` is what you expect?