ros2/rclcpp #1506: Make the lifecycle services optional

Karsten1987, 2021-01-06:
There is currently no way to not have services attached to the lifecycle node. That prevents an easy setup of a lifecycle management system as each loaded lifecycle node can be individually controlled and thus does not have to go through the management system.
This is a precise use case we see at ros control, where each controller is a lifecycle node but should solely be controlled via the controller manager - not the controller itself - in order to guarantee a proper cleanup and resource management.

see https://github.com/ros-controls/ros2_control/issues/266

Karsten1987, 2021-01-06:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2021-03-02:
CI once more:

* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2021-03-16:
@fujitatomoya @clalancette What's the final verdict here? We'd like to get this in before the API freeze in order to move forward with ros-control in Galactic. 
It looks like the rcl related PR (ros2/rcl#882) got approved, the rclcpp PR (#1507) is lacking an approval/review.

fujitatomoya, 2021-03-16:
@Karsten1987 

i will re-visit https://github.com/ros2/rclcpp/pull/1507 in today.

we could use at least another review for both, i guess.

Karsten1987, 2021-03-19:
one more CI after rebase:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2021-03-22:
With state machine options struct:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2021-03-23:
(hopefully) the last round before merging:
* Linux [image]
* Linux-aarch64 [image]
* macOS [image]
* Windows [image]

Karsten1987, 2021-03-23:
closing with #1507 and https://github.com/ros2/rcl/pull/882

closed 2021-03-23 by Karsten1987