ros2/rclcpp #1519: terminate called after throwing an instance of 'std::system_error'   what():  Resource deadlock avoided

LouisChen1905, 2021-01-19:
## Bug report

**Required Info:**

- Operating System:
  - ubuntu 20.04-->
- Installation type:
  - from source
- Version or commit hash:
  - eloquent
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I'm writing gtest scripts. My SetUp():
```c++
auto context = contexts::default_context::get_global_default_context();
        auto options = NodeOptions()
                .context(context)
            // BUG: if set true, test would crash: std::system_error Resource deadlock avoided!
                .use_intra_process_comms(true)
                .automatically_declare_parameters_from_overrides(true);
        mappingNode_ = std::make_shared<Node>("mapping", options);
        bool res = false;
        res = Mapping::getInstance()->initialize(mappingNode_);
        if(!res){
            RCLCPP_ERROR(mappingNode_->get_logger(), "Failed to initialize.");
        }

        executor = std::make_shared<executors::MultiThreadedExecutor>
                   (executor::ExecutorArgs(), 2);
        executor->add_node(mappingNode_);
```

#### Expected behavior

#### Actual behavior

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

clalancette, 2021-01-19:
Are you really using Eloquent on Ubuntu 20.04?  That isn't a supported combination (and in point of fact, Eloquent is end-of-life now anyway).  Can you try this on Foxy on Ubuntu 20.04 and see if you are still running into problems?

LouisChen1905, 2021-01-21:
> Are you really using Eloquent on Ubuntu 20.04? That isn't a supported combination (and in point of fact, Eloquent is end-of-life now anyway). Can you try this on Foxy on Ubuntu 20.04 and see if you are still running into problems?

iuhilnehc-ynos, 2021-01-21:
@LouisChen1905 

I am interested in this issue.
Could you provide a sample or test that can be built and run to reproduce this problem?

BTW: 
```
#14 0x00007ffff7af2d3e in rclcpp::experimental::IntraProcessManager::remove_subscription (this=0x55555617aea0, 
    intra_process_subscription_id=2)
    at /home/chervon/ros2_eloquent_source/src/ros2/rclcpp/rclcpp/src/rclcpp/intra_process_manager.cpp:86
```
Did you change the `rclcpp` source code? (https://github.com/ros2/rclcpp/blob/eloquent/rclcpp/src/rclcpp/intra_process_manager.cpp#L83-L87)

About `std::system_error Resource deadlock avoided!`, it seems there are two places to call `std::unique_lock<std::shared_timed_mutex> lock(mutex_);` in one thread.

LouisChen1905, 2021-01-21:
@iuhilnehc-ynos 

I’ll write a sample program and upload soon.

LouisChen1905, 2021-01-21:
@iuhilnehc-ynos 

> Did you change the rclcpp source code?
I didn't change the source code.

Finally, I found the way to reproduce the error. If I implement the callback of subscription in the `TestClass`, the program runs ok. If I implement the callback in another class `Module1` whose instance is a member of `TestClass`, the error is reproduced. 
```C++
class TestClass
{
public:
  static std::shared_ptr<TestClass> getInstance(){
    static std::shared_ptr<TestClass> instance;
    if(!instance)
      instance.reset(new TestClass());
    return instance;
  }

  bool init(rclcpp::Node::SharedPtr node){
    node_ = node;

    rclcpp::Parameter param;
    node_->get_parameter("topic1", param);
    std::string topic = param.get_value<std::string>();
    RCLCPP_INFO_STREAM(node_->get_logger(), "read topic1: " <<topic);

    module1 = std::make_shared<Module1>(node_);
    //NOTE: uncomment this line to reproduce the bug
//    auto fcn = std::bind(&Module1::stringCallback, module1, std::placeholders::_1);
    // run ok
    auto fcn = std::bind(&TestClass::stringCallback, this, std::placeholders::_1);
    stringSub_ = node_->create_subscription<std_msgs::msg::String>(
          "/string",
          1,
          fcn);

    int16Pub_ = node_->create_publisher<std_msgs::msg::Int16>(
          "/int",1);
    auto pubCB = [this](){
      std_msgs::msg::Int16 msg;
      msg.set__data(1);
      int16Pub_->publish(msg);
    };

    pubTimer_ = node_->create_wall_timer(10ms, pubCB);

    return true;
  }

  void stringCallback(std_msgs::msg::String::SharedPtr msg){
//    RCLCPP_INFO(node_->get_logger(), "received message");
  }

protected:
  TestClass();
private:
  rclcpp::Node::SharedPtr node_;

  rclcpp::Subscription<std_msgs::msg::String>::SharedPtr stringSub_;

  rclcpp::Publisher<std_msgs::msg::Int16>::SharedPtr int16Pub_;
  rclcpp::TimerBase::SharedPtr pubTimer_;

  std::shared_ptr<Module1> module1;
};
```
The project is upload in the attached zip file.
[bug_test.zip](https://github.com/ros2/rclcpp/files/5848820/bug_test.zip)

LouisChen1905, 2021-01-21:
The problem also has something to do with the singleton implementation of `TestClass`. When `TestClass` is written like:
```C++
class TestClass
{
public:
/*  static std::shared_ptr<TestClass> getInstance(){
    static std::shared_ptr<TestClass> instance;
    if(!instance)
      instance.reset(new TestClass());
    return instance;
  }*/

  bool init(rclcpp::Node::SharedPtr node){
    node_ = node;

    rclcpp::Parameter param;
    node_->get_parameter("topic1", param);
    std::string topic = param.get_value<std::string>();
    RCLCPP_INFO_STREAM(node_->get_logger(), "read topic1: " <<topic);

    module1 = std::make_shared<Module1>(node_);
    //NOTE: uncomment this line to reproduce the bug
    auto fcn = std::bind(&Module1::stringCallback, module1, std::placeholders::_1);
    // run ok
//    auto fcn = std::bind(&TestClass::stringCallback, this, std::placeholders::_1);
    stringSub_ = node_->create_subscription<std_msgs::msg::String>(
          "/string",
          1,
          fcn);

    int16Pub_ = node_->create_publisher<std_msgs::msg::Int16>(
          "/int",1);
    auto pubCB = [this](){
      std_msgs::msg::Int16 msg;
      msg.set__data(1);
      int16Pub_->publish(msg);
    };

    pubTimer_ = node_->create_wall_timer(10ms, pubCB);

    return true;
  }

  void stringCallback(std_msgs::msg::String::SharedPtr msg){
//    RCLCPP_INFO(node_->get_logger(), "received message");
  }

  public:
  TestClass();
private:
  rclcpp::Node::SharedPtr node_;

  rclcpp::Subscription<std_msgs::msg::String>::SharedPtr stringSub_;

  rclcpp::Publisher<std_msgs::msg::Int16>::SharedPtr int16Pub_;
  rclcpp::TimerBase::SharedPtr pubTimer_;

  std::shared_ptr<Module1> module1;
};
```
And SetUp() is:
```C++
void SetUp(){
    auto context = contexts::default_context::get_global_default_context();
    auto options = NodeOptions()
            .context(context)
        // BUG: if set true, test would crash: std::system_error Resource deadlock avoided!
            .use_intra_process_comms(true)
            .automatically_declare_parameters_from_overrides(true);
    node_ = std::make_shared<Node>("test_bug", options);

    // NOTE:uncomment this line to reproduce bug
//    TestClass::getInstance()->init(node_);

    testClass_ = std::make_shared<TestClass>();
    testClass_->init(node_);

    pubNode_ = std::make_shared<Node>("pub_node", options);
    stringPub_ = pubNode_->create_publisher<std_msgs::msg::String>("/string", 1);
    auto pubTimerCB = [this](){
      std_msgs::msg::String msg;
      msg.set__data("test");
      stringPub_->publish(std::move(msg));
    };
    pubTimer_ = pubNode_->create_wall_timer(100ms, pubTimerCB);

//    auto intCallback = [this](std_msgs::msg::Int16::SharedPtr msg){
//      RCLCPP_INFO(pubNode_->get_logger(), "Receive int message");
//    };
//    intSub_ = pubNode_->create_subscription<Int16>("/int", 1, intCallback);

    executor_ = std::make_shared<executors::MultiThreadedExecutor>();
    executor_->add_node(node_);
    executor_->add_node(pubNode_);
  }
```
the test program runs OK.

iuhilnehc-ynos, 2021-01-22:
I don't want to argue about whether the design of the sample is reasonable or not,
but I think `rclcpp` should not make users' application aborted no matter how user application to use this library.

We can find that

https://github.com/ros2/rclcpp/blob/b1ff2d5bdc9a7fcc2c0e2aeee0216ebe8d22751d/rclcpp/src/rclcpp/intra_process_manager.cpp#L83-L85
```cpp
  subscriptions_.erase(intra_process_subscription_id);
  ^^^^^^^^^^^^^^^^^^^^ // There exists a case to destroy
                       // `AnySubscriptionCallback any_callback_` of SubscriptionIntraProcess
                       //, which might keep the lifecycle of node that will call `IntraProcessManager::remove_subscription`.
```

How about adding a new `recusive_mutex` for `subscriptions_`? @ivanpauno , do you have any suggestions?

fujitatomoya, 2021-01-22:
@iuhilnehc-ynos 

> How about adding a new recusive_mutex for subscriptions_?

could you elaborate why adding recursive_mutex is gonna address this issue?

the root cause of this is dead lock as following,

```
#9  0x00007ffff6b0f77f in std::__throw_system_error(int) () from /lib/x86_64-linux-gnu/libstdc++.so.6
#10 0x00007ffff7af3a99 in std::__shared_mutex_pthread::lock (this=0x55555617af50) at /usr/include/c++/9/shared_mutex:188
#11 0x00007ffff7af3b34 in std::shared_timed_mutex::lock (this=0x55555617af50) at /usr/include/c++/9/shared_mutex:459
#12 0x00007ffff7af519b in std::unique_lock<std::shared_timed_mutex>::lock (this=0x7ffffffea490)
    at /usr/include/c++/9/bits/unique_lock.h:141
#13 0x00007ffff7af45dd in std::unique_lock<std::shared_timed_mutex>::unique_lock (this=0x7ffffffea490, __m=...)
    at /usr/include/c++/9/bits/unique_lock.h:71
#14 0x00007ffff7af2d3e in rclcpp::experimental::IntraProcessManager::remove_subscription (this=0x55555617aea0, 
    intra_process_subscription_id=2)
    at /home/chervon/ros2_eloquent_source/src/ros2/rclcpp/rclcpp/src/rclcpp/intra_process_manager.cpp:86
#15 0x00007ffff7bd3b67 in rclcpp::SubscriptionBase::~SubscriptionBase (this=0x55555619ac90, __in_chrg=<optimized out>)
    at /home/chervon/ros2_eloquent_source/src/ros2/rclcpp/rclcpp/src/rclcpp/subscription_base.cpp:96
```

https://github.com/gcc-mirror/gcc/blob/e90bb02d5ba3853b4abb0c66f3e9a41bd03a4294/libstdc%2B%2B-v3/include/std/shared_mutex#L186-L188

trying to rwlock with writer access, it detects EDEADLK from glibc then raise exception.

iuhilnehc-ynos, 2021-01-22:
@fujitatomoya 

> trying to rwlock with writer access, it detects EDEADLK from glibc then raise exception.

About std::system_error Resource deadlock avoided!, it seems there are two places to call std::unique_lock<std::shared_timed_mutex> lock(mutex_); in one thread. (As I mentioned at https://github.com/ros2/rclcpp/issues/1519#issuecomment-764467045)

```cpp
a thread to run:
{
 std::unique_lock<std::shared_timed_mutex> lock(mutex_); 
 subscriptions_.erase(intra_process_subscription_id);   ==> if `erase` lead to run `std::unique_lock<std::shared_timed_mutex> lock(mutex_);` again, it throws exception(`EDEADLK`).
}
```

which mutex do you think can fix this kind of error? From what I know, `recursive_mutext` is a better choice.

fujitatomoya, 2021-01-22:
> if `erase` lead to run `std::unique_lock<std::shared_timed_mutex> lock(mutex_);` again, it throws exception(`EDEADLK`).

i think in that case, `device_or_resource_busy (EBUSY)` should be seen on stack trace because that is not a deadlock?

iuhilnehc-ynos, 2021-01-22:
@fujitatomoya 

> i think in that case, `device_or_resource_busy (EBUSY)` should be seen on stack trace because that is not a deadlock?

Thank you for your suggestion.

I think `lock a mutex twice` is also a simple type of deadlock,
so calling `std::unique_lock<std::shared_timed_mutex> lock(mutex_);` twice to throw EDEADLK is acceptable.

Based on your suggestion about throws exception in this case, so it's the users' responsibility to catch the exception.
I'll wait to hear others' opinions.

LouisChen1905, 2021-01-24:
@iluetkeb 
Is it better for  the output behaviour  of `rclcpp`for users to be  independent of the configuration input by  `NodeOptions`. Since if `.use_intra_process_comms(true)` the code chrashes. If parameter is `false`, it runs without crashing. AFAK, the differencee between them is the different communicaton mechanisms leading to higher or lower efficiency. If one the options makes user's program crash, while the other does not, the user may get confused.

iluetkeb, 2021-01-25:
@LouisChen1905 I think you tagged me by mistake, right? I haven't been involved in this ticket before.

iuhilnehc-ynos, 2021-01-28:
- refer to https://github.com/ros2/rclcpp/issues/1519#issuecomment-764560425

```
$ git clone https://github.com/iuhilnehc-ynos/ros2_issues_test.git
$ cd ros2_issues_test
$ source /opt/ros/foxy/setup.bash      ## or use ros2 master workspace
$ colcon build --packages-select rclcpp_issue_1519       
$ source install/setup.bash
$ ./build/rclcpp_issue_1519/test1519
```

- PR to fix this issue

https://github.com/ros2/rclcpp/pull/1530

closed 2021-02-03 by iuhilnehc-ynos

iuhilnehc-ynos, 2021-02-03:
@LouisChen1905

Could you run your sample with `master` branch, at your convenience? Consider this PR breaking the ABI, I'll not provide a patch for `eloquent` or `foxy`.

If you have any questions, please reopen this. Thanks.

PR #1530 (2021-02-03): Avoid `Resource deadlock avoided` if use intra_process_comms
Fixes #1519 

NOTE: To reproduce the issue, please refer to https://github.com/ros2/rclcpp/issues/1519#issuecomment-768817649.