ros2/rclcpp #1523: 👩‍🌾 test_graph_services_by_node regression

chapulina, 2021-01-22:
## Bug report

**Required Info:**

- Operating System:
  - Linux
- Installation type:
  - source
- Version or commit hash:
  - <master
- DDS implementation:
  - default
- Client library (if applicable):
  - `rclcpp`

#### Steps to reproduce issue

Test `rclcpp_lifecycle.TestDefaultStateMachine.test_graph_services_by_node` failed for the first time in nightly CI on 2021/01/21.

See [test history](https://ci.ros2.org/job/nightly_linux_release/1802/testReport/rclcpp_lifecycle/TestDefaultStateMachine/test_graph_services_by_node/history/).

#### Expected behavior

Test passes

#### Actual behavior

Test failed

#### Additional information

Failure output:

```
/home/jenkins-agent/workspace/nightly_linux_release/ws/src/ros2/rclcpp/rclcpp_lifecycle/test/test_lifecycle_node.cpp:629
Expected: (service_names_and_types_by_node.end()) != (service_names_and_types_by_node.find(std::string("/testnode/change_state"))), actual: 8-byte object <18-87 D4-21 FE-7F 00-00> vs 8-byte object <18-87 D4-21 FE-7F 00-00>
```

---

https://github.com/osrf/buildfarmer/issues/145

clalancette, 2021-02-18:
Well, we haven't seen this fail since that one time (almost a month ago).  It might be an extreme flake, but I'm just going to close it out for now and we can revisit it if it comes back.

closed 2021-02-18 by clalancette

jacobperron, 2021-03-22:
I've seen this failure locally and in some PR jobs, e.g. https://build.ros2.org/job/Rpr__rclcpp__ubuntu_focal_amd64/1022/