ros2/rclcpp #1524: rclcpp::spin_some(node)  Occasionally crash.

BaiCai1990, 2021-01-23:
rclcpp::WallRate main_loop(100);     
while(rclcpp::ok)
{
  rclcpp::spin_some(ros_node_);
  current_state = GetCurrentState();
  switch(current_state){
  case IDLE:{
    rclcpp::WallRate r(100);
    while(rclcpp::ok() && GetCurrentState() == IDLE){
      rclcpp::spin_some(ros_node_);                               //!!!!!!!!!!!!!!!!!!!!!!!! **crashed here**!!!!
      // other code
      r.sleep();
    }
    break;
  }
  case WALK:{
    rclcpp::WallRate r(100);
    while(rclcpp::ok() && GetCurrentState() == WALK){
      rclcpp::spin_some(ros_node_);
      // other code
      r.sleep();
    }
    break;
  }

  main_loop.sleep();
}

//!================================================
Above is my program, it has two state, IDLE and WALK. 
When state switch from WALK to IDLE, the program will  **occasionally** crash in rclcpp::spin_some(ros_node_), 
My ros version is Foxy.
In WALK state the loop is time-consumming than IDLE state.

wjwwood, 2021-01-23:
Please use the issue template:

https://raw.githubusercontent.com/ros2/rclcpp/master/.github/ISSUE_TEMPLATE.md

It should have been pre-filled into your description when creating a new issue.

Also, please provide a traceback if you can. Though not specific to `rclcpp` the ROS Nav2 project has a nice tutorial for how to get this:

https://navigation.ros.org/tutorials/docs/get_backtrace.html

I'll close this until it's fixed up. Please feel free to edit it or comment here, and I can reopen it in the future.

BaiCai1990, 2021-01-23:
Done.
Thinks for your attention.

clalancette, 2021-02-11:
@BaiCai1990 The information provided is more helpful (it shows us that it is down in the Fast-RTPS layer).  But it is much easier for us to debug if you can provide a completely compiling, working example.  Could you please do that so we can try to assign it to the right people?

ivanpauno, 2021-02-17:
@BaiCai1990 please also provide a multithreaded traceback (`thread apply all bt` in gdb), and what exactly was the error message and return code when crashing.

DensoADAS, 2021-04-15:
perhaps this is related to https://github.com/ros2/rclcpp/pull/1632 ?
I had at least a similar issue and solved it for me like this

clalancette, 2022-02-03:
No response in almost a year, so closing.  Please feel free to provide the requested information and reopen.

closed 2022-02-03 by clalancette