ros2/rclcpp #1525: [Bug?] warning messages when set parameters through rclcpp::NodeOptions().arguments() function

hyunseok-yang, 2021-01-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 20.04
- Installation type:
  - binary foxy latest version
- Version or commit hash:
  - dpkg -l | grep ros-foxy-rclcpp  
  - ii  ros-foxy-rclcpp   **2.3.0-1focal.20201228.140355**          amd64        The ROS client library in C++
- DDS implementation:
  - Fast RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

I created my own project and coded one derived class like below.

```c++
Base::Base(const string node_name_, const string namespace_, const rclcpp::NodeOptions &options_, const int number_of_bridges)
    : Node(node_name_,
           namespace_,
           rclcpp::NodeOptions(options_)
               .arguments(vector<string>{"/tf:=tf", "/tf_static:=tf_static"})
               .append_parameter_override("use_sim_time", true)
               .automatically_declare_parameters_from_overrides(true))
```

Then it shows warning message like this, but it works well as I intend.
```shell
[WARN 1611508484.187430663] [rcl]: Found remap rule '/tf:=tf'. This syntax is deprecated. Use '--ros-args --remap /tf:=tf' instead. (rcl_parse_arguments() at /tmp/binarydeb/ros-foxy-rcl-1.1.10/src/rcl/arguments.c:558)
[WARN 1611508484.187536111] [rcl]: Found remap rule '/tf_static:=tf_static'. This syntax is deprecated. Use '--ros-args --remap /tf_static:=tf_static' instead. (rcl_parse_arguments() at /tmp/binarydeb/ros-foxy-rcl-1.1.10/src/rcl/arguments.c:558)
```

So I tried to change parameters to suppress warning message.

```c++
.arguments(vector<string>{"--ros-args --remap /tf:=tf", '--ros-args --remap /tf_static:=tf_static"})
```

#### Expected behavior

remapping should be worked!

#### Actual behavior

nothing happened with this.
```c++
.arguments(vector<string>{"--ros-args --remap /tf:=tf", '--ros-args --remap /tf_static:=tf_static"})
```

#### Additional information

Anyone tried this? 
I think that should be worked.

fujitatomoya, 2021-01-25:
No, this is not a bug.

as you see warnings, the syntax has been deprecated. `--ros-args` flag is introduced on Eloquent and later, you can refer to https://index.ros.org/doc/ros2/Tutorials/Node-arguments/.

fujitatomoya, 2021-01-25:
@hyunseok-yang 

if you don't have further question, i think we can close this one.

iuhilnehc-ynos, 2021-01-25:
> .arguments(vector<string>{"--ros-args --remap /tf:=tf", '--ros-args --remap /tf_static:=tf_static"})

It seems a wrong format, please use the following,

```
 .arguments(vector<string>{"--ros-args",  "--remap", "/tf:=tf", "--remap", "/tf_static:=tf_static"})
```
or
``` 
 .arguments({"--ros-args",  "-r", "/tf:=tf", "-r", "/tf_static:=tf_static"})
```

hyunseok-yang, 2021-01-25:
@fujitatomoya 
yes, I know the deprecated syntax. @iuhilnehc-ynos  gave me a solution like above :)

I used wrong syntax in .arguments() api.

This format is working good! perfect :+1: 
```c++
 .arguments(vector<string>{"--ros-args",  "--remap", "/tf:=tf", "--remap", "/tf_static:=tf_static"})
```

closed 2021-01-25 by hyunseok-yang

fujitatomoya, 2021-01-25:
@hyunseok-yang Yeah 😄 i was too early to reply, my bad sorry about that! anyway @iuhilnehc-ynos thanks for the follow-up 👍