ros2/rclcpp #1539: Foxy Flaky test: rclcpp_action.TestClientAgainstServer.execute_rcl_errors Fails: error taking result response

sloretz, 2021-02-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu Focal
- Installation type:
  - Source
- Version or commit hash:
  - 6756ccfb50590225c8f24d9c8b96d0598a124093
- DDS implementation:
  - Connext, CycloneDDS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
colcon test --event-handlers console_direct+ --packages-select rclcpp_action --ctest-args -R test_client --repeat-until-fail 1000
```

#### Expected behavior
Test always passes

#### Actual behavior

Occasionally fails with:

```
1: [ RUN      ] TestClientAgainstServer.execute_rcl_errors
1: unknown file: Failure                              
1: C++ exception with description "error taking result response: error not set" thrown in the test body.
1: [  FAILED  ] TestClientAgainstServer.execute_rcl_errors (89 ms)
1: [----------] 1 test from TestClientAgainstServer (89 ms total)
```

#### Additional information

Failed in this job:

http://build.ros2.org/job/Fci__nightly-connext_ubuntu_focal_amd64/231

I've so far not seeing it fail while running the test under `gdb` or `rr`. It does not seem to be related to the cleanup of other tests because I've seen it fail with all other tests in `test_client` besides `execute_rcl_errors` commented out.

Flaky test tracker: osrf/buildfarmer#145

sloretz, 2021-02-03:
I am able to reproduce it with colcon `--retest-until-fail` with a large number `>500`, but so far unable to reproduce with gdb.

Currently trying:

```bash
$  gdb /home/sloretz/bigssd/foxy/build/rclcpp_action/test_client
```

then
```gdb
set pagination off
b _exit
commands
r
end
b client.c:318
r
```

Where `client.c:318` is: https://github.com/ros2/rcl/blob/7ddea6c155fc4e6b4331f8c0de964f6e6f673cb1/rcl/src/rcl/client.c#L318

sloretz, 2021-02-05:
I was unable to reproduce in gdb or `rr record --chaos`, but am able to reproduce it on it's own with `--retest-until-fail`

iuhilnehc-ynos, 2021-06-24:
I can reproduce this issue with `latest foxy` using `rmw_connext_cpp` and `rmw_fastrtps_cpp`, but I can't reproduce it with `rmw_cyclonedds_cpp` for now. ( NOTE: reproduced it with `master` using `rmw_fastrtps_cpp` also.)

<details><summary> log </summary>
<p>

```shell
- rmw_fastrtps_cpp

NOTE: easy to reproduce the failure by using rmw_fastrtps_cpp,
please use a shell script to run loop with command `./test_client --gtest_filter=*execute_rcl_errors`

Note: Google Test filter = *execute_rcl_errors
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from TestClientAgainstServer
[ RUN      ] TestClientAgainstServer.execute_rcl_errors
unknown file: Failure
C++ exception with description "error taking result response: error not set" thrown in the test body.
[  FAILED  ] TestClientAgainstServer.execute_rcl_errors (40 ms)
[----------] 1 test from TestClientAgainstServer (40 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (41 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestClientAgainstServer.execute_rcl_errors

- rmw_connext_cpp

Note: Google Test filter = *execute_rcl_errors
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from TestClientAgainstServer
[ RUN      ] TestClientAgainstServer.execute_rcl_errors
RTI Data Distribution Service Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
RTI Data Distribution Service Non-commercial license is for academic, research, evaluation and personal use only. USE FOR COMMERCIAL PURPOSES IS PROHIBITED. See RTI_LICENSE.TXT for terms. Download free tools at rti.com/ncl. License issued to Non-Commercial User license@rti.com For non-production use only.
Expires on 00-jan-00 See www.rti.com for more information.
unknown file: Failure
C++ exception with description "error taking result response: error not set" thrown in the test body.
[  FAILED  ] TestClientAgainstServer.execute_rcl_errors (7430 ms)
[----------] 1 test from TestClientAgainstServer (7430 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (7440 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] TestClientAgainstServer.execute_rcl_errors

```
</p>
</details>

I found that the the `result_response` message requested from [the previous action_client->async_get_result](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L918) will cause [the latter dual_spin_until_future_complete](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L941) to throw an exception, because the `rcl_action_take_result_response` is [patched before](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L937-L938).

FYI: Comment [L942-L946](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L942-L946) of  and add `assert(flase)` [here](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/src/client.cpp#L427) to genereate coredump to confirm the backtrace of exception
<details><summary> backtrace </summary>
<p>

```shell
(gdb) bt
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:50
#1  0x00007fc88b16a859 in __GI_abort () at abort.c:79
#2  0x00007fc88b16a729 in __assert_fail_base (fmt=0x7fc88b300588 "%s%s%s:%u: %s%sAssertion `%s' failed.\n%n", assertion=0x7fc88be67af2 "false", 
    file=0x7fc88be67488 "/home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp_action/src/client.cpp", line=427, function=<optimized out>)
    at assert.c:92
#3  0x00007fc88b17bf36 in __GI___assert_fail (assertion=0x7fc88be67af2 "false", 
    file=0x7fc88be67488 "/home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp_action/src/client.cpp", line=427, 
    function=0x7fc88be67ac0 "virtual void rclcpp_action::ClientBase::execute()") at assert.c:101
#4  0x00007fc88be4d5e3 in rclcpp_action::ClientBase::execute (this=0x563803efeba0)
    at /home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp_action/src/client.cpp:427
#5  0x00007fc88bb5c18b in rclcpp::Executor::execute_any_executable (this=0x563803c25508, any_exec=...)
    at /home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:316
#6  0x00007fc88bb5ba95 in rclcpp::Executor::spin_some (this=0x563803c25508, max_duration=...)
    at /home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp/src/rclcpp/executor.cpp:247
#7  0x0000563802886999 in TestClient::dual_spin_until_future_complete<std::shared_ptr<rclcpp_action::ClientGoalHandle<test_msgs::action::Fibonacci> > >
    (this=0x563803c253b0, future=...) at /home/chenlh/Projects/ROS2/rci_nighty_ws/src/ros2/rclcpp/rclcpp_action/test/test_client.cpp:255
#8  0x000056380286f568 in TestClientAgainstServer_execute_rcl_errors_Test::TestBody (this=0x563803c253b0)
[... cut ...]
It seems that we can't expect [spin_once](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L129-L130) to make the [exception thrown](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L920) immediately, which means the following messages, such as [result_response](https://github.com/ros2/rclcpp/blob/1fff1b7cbafcbc698b5dda5b8e4a35a27c6748c4/rclcpp_action/test/test_client.cpp#L148) message, could be published.

About the fix, @hidmic @brawner, do you have any suggestions?

clalancette, 2024-01-19:
We haven't seen this particular issue in a long time, so I'll close this out.  If we do see it in CI again, we can open up a new bug.

closed 2024-01-19 by clalancette