ros2/rclcpp #1540: Unify clock behavior between use_sim_time true/false

doisyg, 2021-02-03:
## Feature request
It is my understanding that one can get very different behavior when asking for the node time when `use_sim_time` is `true` and the simulation time is published on the `/clock` topic, compared to when it is `false`.
See for instance this code that blocks forever in the timer_callback when `use_sim_time` is `true` and that behave properly when `use_sim_time` is `false`:

```
#include <rclcpp/rclcpp.hpp>
#include <chrono>
#include <std_msgs/msg/string.hpp>

using namespace std::chrono_literals;

class TimeTest: public rclcpp::Node
{
  public:
    TimeTest()
      : Node("time_test_node")
    {
      this->set_parameter(rclcpp::Parameter("use_sim_time", true));
      timer = this->create_wall_timer(2s, std::bind(&TimeTest::timer_callback, this));
    }

  private:
    rclcpp::TimerBase::SharedPtr timer;

    void timer_callback()
    {
      rclcpp::Time start_time = this->get_clock()->now();
      bool is_time_out = false;
      while(!is_time_out){
        rclcpp::Time time_now = this->get_clock()->now();
        rclcpp::Duration time_spent = time_now -start_time;
        RCLCPP_INFO_STREAM(this->get_logger(), " now (s): " << time_now.seconds());
        rclcpp::sleep_for(100ms);
        is_time_out = time_spent.seconds() > 1.0;
      }
    }
};

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<TimeTest>());
  rclcpp::shutdown();
  return 0;
}
```

It would be really useful to have the same behavior in both case.

#### Feature description
The node clock should be updated in a similar way with `use_sim_time` `true`/`false`, i.e no clock freeze specific to `use_sim_time` `true`

#### Implementation considerations
I guess the clock freeze `with use_sim_time` is `true` in callbacks because a node is by default mono-threaded and hence the simulated clock cannot be updated (the internal callback of the `/clock` topic cannot be served in parallel of another callback). 
One solution could be to add a specific thread to the node to serve the `/clock` topic and update the node clock when `use_sim_time` is `true` (in a transparent way for the user of the node).

wjwwood, 2021-02-04:
I think this should be "won't fix", as you shouldn't be blocking in a callback.

The reason it blocks is that you're using a single threaded executor (default for `rclcpp::spin()`), and it could be fixed by using a multi-threaded executor, but a well designed node should not depend on that.

If you need to have long running callbacks, I'd recommend instead having your own thread that can run those long blocking tasks so they don't interfere with other tasks that the node has to do.

For example something like this:

```c++
#include <rclcpp/rclcpp.hpp>
#include <chrono>
#include <std_msgs/msg/string.hpp>

using namespace std::chrono_literals;

class TimeTest: public rclcpp::Node
{
  public:
    TimeTest()
      : Node("time_test_node")
    {
      this->set_parameter(rclcpp::Parameter("use_sim_time", true));
      my_callback_group = this->create_callback_group(rclcpp::CallbackGroupType::MutuallyExclusive, false);
      timer = this->create_wall_timer(2s, std::bind(&TimeTest::timer_callback, this), my_callback_group);
      my_executor_thread = std::thread([this]() {
        this->my_executor.add_callback_group(this->my_callback_group, this->get_node_base_interface());
        this->my_executor.spin();
      });
    }

    ~TimeTest()
    {
      this->my_executor.cancel();
      this->my_executor_thread.join();
    }

  private:
    rclcpp::CallbackGroup::SharedPtr my_callback_group;
    rclcpp::executors::SingleThreadedExecutor my_executor;
    std::thread my_executor_thread;
    rclcpp::TimerBase::SharedPtr timer;

    void timer_callback()
    {
      rclcpp::Time start_time = this->get_clock()->now();
      bool is_time_out = false;
      while(!is_time_out){
        rclcpp::Time time_now = this->get_clock()->now();
        rclcpp::Duration time_spent = time_now -start_time;
        RCLCPP_INFO_STREAM(this->get_logger(), " now (s): " << time_now.seconds());
        rclcpp::sleep_for(100ms);
        is_time_out = time_spent.seconds() > 1.0;
      }
    }
};

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<TimeTest>());
  rclcpp::shutdown();
  return 0;
}
```

This is a lot more boilerplate, but no more than the boilerplate you'd need to setup in order to have your own thread-pool for handling these tasks and not causing a hangup in the node.

This ability to have your own callback group and executor for individual items in the node is a new feature, and hopefully in time for Galactic we'll have some syntactic sugar to make things like this easier (less code).

wjwwood, 2021-02-04:
It may not seem like it, but this is actually the same issue as https://github.com/ros2/rclcpp/issues/1533, which to generalize it is to "be able to wait on asynchronous things within a ROS callback safely". In this issue it's waiting on the message on the `/clock` topic to be handled and in the other issue it is waiting on the response from the service server to be received.

I'm actually gonna close this and say let's focus discussion there.

If instead, you think the `/clock` should insulate itself (maybe always create its own thread or something), then I'd say let's open a new issue instead.

closed 2021-02-04 by wjwwood

doisyg, 2021-02-04:
Thanks for your detailed answer, I thought of using CallbackGroup to solve this issue and your example is very useful to show how to do it properly. And I agree and also identified the same issue as https://github.com/ros2/rclcpp/issues/1533
However I will open a new issue on the insulation of the `/clock` : https://github.com/ros2/rclcpp/issues/1542

doisyg, 2021-02-04:
> This ability to have your own callback group and executor for individual items in the node is a new feature, and hopefully in time for Galactic we'll have some syntactic sugar to make things like this easier (less code).

add_callback_group is not part in foxy, would it be possible to port it ?

wjwwood, 2021-02-05:
> add_callback_group is not part in foxy, would it be possible to port it ?

Hmm, I don't think so. I could check again, but I think the changes are not easily modified to be ABI compatible with Foxy (the requirement for a back port). You'll have to either get it from source or wait for Galactic I think.

doisyg, 2021-02-05:
I compiled from source from foxy branch + cherry pick of the commit adding this functionality and it works (but would be amazing if you find a way of backporting it at some point)