ros2/rclcpp #1542: Insulate /clock callback on specific thread

doisyg, 2021-02-04:
## Feature request
As discussed here: https://github.com/ros2/rclcpp/issues/1540. By design, and in the words of @wjwwood , by default it is NOT possible to "wait on asynchronous things within a ROS callback safely", see https://github.com/ros2/rclcpp/issues/1533
The issue (or more accurately behavior by design) is well know. 
However with the current implementation of simulation time which listen to the `/clock` topic, this behavior breaks the symmetry  between the usage of real time (system time internally accessed without the need of subscribing to a topic, and hence always updated) and simulation time (time received on the `/clock `topic and hence frozen in a callback).
I believe the symmetry should be maintain in order to not draw false conclusions about the behavior of a code developed in simulation VS real time, and thus independently from good/bad code practice.

#### Implementation considerations
Maybe insulation the `/clock` subscriber callback on a specific thread ?

clalancette, 2021-02-18:
Assigning to one of the rclcpp maintainers to decide what to do with this feature request; whether to put it on the backlog, help-wanted, or to close it out.  Once we have figured out the status, feel free to unassign yourself.

fujitatomoya, 2021-02-19:
@doisyg could you elaborate a little bit?

> by default it is NOT possible to "wait on asynchronous things within a ROS callback safely"

isn't that supposed to be `synchronous`? if im not mistaken.

> time received on the /clock topic and hence frozen in a callback

this means that clock topic is being published certain frequency, right? but it still gets stuck on subscription callback on topic `/clock`?

doisyg, 2021-02-19:
> > by default it is NOT possible to "wait on asynchronous things within a ROS callback safely"
> 
> isn't that supposed to be `synchronous`? if im not mistaken.

These are @wjwwood words, but I believe asynchronous is the right term for anything going through topics.

> this means that clock topic is being published certain frequency, right? but it still gets stuck on subscription callback on topic `/clock`?

Exactly, when `use_sim_time` is `true`,  it gets stuck on the subscription callback which is not executed while any other callback is served. Hence a node clock is not having the same update behavior from the user perspective when `use_sim_time` is `true` versus `use_sim_time` is `false` (where getting an updated system time is immediate and completely from thread mechanisms).

Executing this sample code with `use_sim_time` is `true` + something publising on the /clock topic VS `use_sim_time` is `false` should highlight the issue:
```
#include <rclcpp/rclcpp.hpp>
#include <chrono>
#include <std_msgs/msg/string.hpp>

using namespace std::chrono_literals;

class TimeTest: public rclcpp::Node
{
  public:
    TimeTest()
      : Node("time_test_node")
    {
      this->set_parameter(rclcpp::Parameter("use_sim_time", true));
      timer = this->create_wall_timer(2s, std::bind(&TimeTest::timer_callback, this));
    }

  private:
    rclcpp::TimerBase::SharedPtr timer;

    void timer_callback()
    {
      rclcpp::Time start_time = this->get_clock()->now();
      bool is_time_out = false;
      while(!is_time_out){
        rclcpp::Time time_now = this->get_clock()->now();
        rclcpp::Duration time_spent = time_now -start_time;
        RCLCPP_INFO_STREAM(this->get_logger(), " now (s): " << time_now.seconds());
        rclcpp::sleep_for(100ms);
        is_time_out = time_spent.seconds() > 1.0;
      }
    }
};

int main(int argc, char **argv)
{
  rclcpp::init(argc, argv);
  rclcpp::spin(std::make_shared<TimeTest>());
  rclcpp::shutdown();
  return 0;
}
```

clalancette, 2021-02-19:
While I will say that the difference between `use_sim_time = True` and `use_sim_time = False` is unfortunate, I will also say that the code you are using as an example is an anti-pattern.  In particular, it is a bad idea to do long sleeps from within callbacks.  That's because in the single-threaded executor case (which is the default), you are preventing *any* other callbacks from running in your process.  There are various ways around this (using the `MultiThreadedExecutor` is one), but in general it is best to do as little work (and no sleeping) as possible inside of the callbacks.

In this case, I would probably rewrite this to use a single-shot timer for the initial 2 second sleep, and then reset the timer period after the first call to 100ms to do the sleeps in between.

doisyg, 2021-02-19:
I definitely agree that the example is an anti-pattern, but I believe the symmetry of behaviors between `use_sim_time = True` and `use_sim_time = False` should be independent from the good / bad coding practice.
The example given is extreme, but nonetheless, without knowing the internal mechanics of how the simulated clock is implemented, it is hard for a naive user to get that the clock that he's using has a behavior that depends that much on the availability of the node thread when ` use_sim_time = True`, whereas when ` use_sim_time = False` it is not

fujitatomoya, 2021-02-20:
thanks for sharing thoughts 👍 

I really do not know this is anti-pattern or not, but just care about application perspective.

> behaviors between use_sim_time = True and use_sim_time = False should be independent from the good / bad coding practice.

this makes sense to me, application does not want to adjust the sequence based on this parameter. (it can do, but does not want to, i guess)

> The example given is extreme, but nonetheless,

would you care to share one actual use case, if you have? i am interested in it.

doisyg, 2021-02-20:
> would you care to share one actual use case, if you have? i am interested in it.

Any accurate usage of the node clock in a callback. I can think of time-stamping sensor data for instance

doisyg, 2021-02-25:
PR here: https://github.com/ros2/rclcpp/pull/1556

fujitatomoya, 2021-03-29:
close this, since https://github.com/ros2/rclcpp/pull/1556 has been merged.

closed 2021-03-29 by fujitatomoya

fujitatomoya, 2021-04-01:
i believe that rclpy has the same problem like this.