ros2/rclcpp #1552: :man_farmer: test_multithreaded_executor is flaky

jacobperron, 2021-02-11:
## Bug report

**Required Info:**

- Operating System:
  - macOS
- Version or commit hash:
  - 09950ba

#### Steps to reproduce issue

Latest failure: https://ci.ros2.org/job/nightly_osx_release/1962/testReport/rclcpp/TestMultiThreadedExecutor/timer_over_take/

Looks like it has been failing for a while now, but not consistently:

https://ci.ros2.org/job/nightly_osx_release/1962/testReport/rclcpp/TestMultiThreadedExecutor/timer_over_take/history

#### Additional information

Looks like the same as https://github.com/ros2/rclcpp/issues/1110, but only noticing the failure on macOS now.

clalancette, 2021-02-12:
I did take a quick look into it a while ago, and I think it is a real bug.  If you look closely at when it fails, the timer is supposed to take *at least* 0.75 seconds.  If you look at the output there, it only took 0.4 seconds, so it is clearly expiring much too early.  macOS has the most "interesting" CPU scheduler of our target platforms, so that is likely why we only see it there.

tfoote, 2021-03-08:
This came up again last night on two separate builds with an even larger underun of 0.22 vs 0.75 seconds. 

`/Users/osrf/jenkins-agent/workspace/nightly_osx_release/ws/src/ros2/rclcpp/rclcpp/test/rclcpp/executors/test_multi_threaded_executor.cpp:91
Expected: (diff) > (PERIOD - TOLERANCE), actual: 0.220429 vs 0.75`

Observed in [OSX nightly repeated](https://ci.ros2.org/job/nightly_osx_repeated/2275/testReport/) and [OSX release](https://ci.ros2.org/job/nightly_osx_release/1990/testReport/)

clalancette, 2021-03-30:
I'm going to close this one as a duplicate of #1008, since we see this on all platforms.

closed 2021-03-30 by clalancette