ros2/rclcpp #1553: SyncParametersClient can't be used in component node

tylerjw, 2021-02-12:
Because `SyncParametersClient` calls `spin_until_future_complete` it can't be used in component nodes.  

A workaround for this might be to have some way of discovering if the node is a component and then just calling `get()` on the future instead of trying to spin the node (as the Node should already be added to an executor and executing).  Another option would be to leave spinning the Node on an executor up to the implementer of the Node (expect they are spinning it already when they call methods like `set_parameters`) and change the implementation of `SyncParametersClient` to block forever (by waiting on the future) if the user failed to start the executor on a separate thread first.

This workaround might not work during initialization if the component manager doesn't start the executor on a separate thread before loading the component nodes.

fujitatomoya, 2021-02-15:
i think this is related to https://github.com/ros2/rclcpp/issues/1533, since ParameterClients are based on services.

fujitatomoya, 2021-02-15:
@tylerjw could you elaborate what you really want to do with more description or sample package that we can build with colcon workspace?

i think [component_container_mt](https://github.com/ros2/rclcpp/blob/master/rclcpp_components/src/component_container_mt.cpp) could solve your problem. to use that with launch file you can change the following into `component_container_mt`.

https://github.com/ros2/demos/blob/11e00ecf7eec25320f950227531119940496d615/composition/launch/composition_demo.launch.py#L28