ros2/rclcpp #1569: goal_handle attempted invalid transition

witignite, 2021-03-04:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: Binaries
- Version or commit hash:
- DDS implementation:
- Client library (if applicable):
- ROS Version: Foxy

#### Issue

It seems that if the function `handle_cancel` (passed to the function [`create_server`](https://docs.ros2.org/foxy/api/rclcpp_action/namespacerclcpp__action.html#aac27305f020b4586d4ae781024ac80e4)) takes some time before returning `rclcpp_action::CancelResponse::ACCEPT` and it returns after the `handle_accepted` has finished (which sets  `goal_handle->succeed(result);`), the following error will happen:
```bash
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  goal_handle attempted invalid transition from state SUCCEEDED with event CANCEL_GOAL, at /tmp/binarydeb/ros-foxy-rcl-action-1.1.10/src/rcl_action/goal_handle.c:95
```
In the code that was tested, `sleep` is put inside `handle_cancel`; in case where the user may want to perform some task before cancelling the goal. Just curious if this is the correct place where the user should perform the "clean-up" task? If not, what is the function `handle_cancel` for?

Another thing is that [this line]() (`goal_handle->publish_feedback(feedback);`) **is not** commented out, the above error won't happen. What could be the reason? Is there any kind of lock mechanism happen under the hood when calling this function?

#### Steps to reproduce issue
In the first terminal:
```bash
ros2 run action_tutorials_cpp fibonacci_action_server
```
In the second terminal:
```bash
ros2 action send_goal /fibonacci action_tutorials_interfaces/action/Fibonacci "{order: 10}"
```
then press Ctrl+C.

#### Expected behavior

#### Actual behavior

fujitatomoya, 2021-03-08:
> It seems that if the function handle_cancel (passed to the function create_server) takes some time before returning rclcpp_action::CancelResponse::ACCEPT and it returns after the handle_accepted has finished (which sets goal_handle->succeed(result);), the following error will happen:

i think this is supposed to happen, because application returns `rclcpp_action::CancelResponse::ACCEPT` after finish the request. i think application has full control of this, here are some examples i can think of.

- if executing prevails to cancellation, `handle_cancel` can check with `goal_handle->is_executing()`, and if that is true, it can return `rclcpp_action::CancelResponse::REJECT`.
- if cancellation prevails to executing, notify `execute` thread and return `rclcpp_action::CancelResponse::ACCEPT`. then `execute` thread can call `goal_handle->canceled()`

fujitatomoya, 2021-03-08:
> However, another thing is that: if this line inside handle_accepted (goal_handle->publish_feedback(feedback);) is not commented out, the above error won't happen; the handle_accepted seems to be blocked and the handle_cancel goes on until finished without an error. What could be the reason? Is there any kind of lock mechanism happen under the hood when calling this line?

i see the same behavior with uncomment [publish_feedback](https://github.com/witignite/demos/blob/100fb00f7290714b260d4c12414f84265301551d/action_tutorials/action_tutorials_cpp/src/fibonacci_action_server.cpp#L120)

```
@@ -117,7 +119,7 @@ private:
       sequence.push_back(sequence[i] + sequence[i - 1]);
       RCLCPP_INFO_STREAM(this->get_logger(), "\tseq ... " << sequence[i] + sequence[i - 1]);
       // Publish feedback
-      // goal_handle->publish_feedback(feedback);
+      goal_handle->publish_feedback(feedback);

       loop_rate.sleep();
     }
```

the same termination as following.

```
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  goal_handle attempted invalid transition from state SUCCEEDED with event CANCEL_GOAL, at /root/ros2_ws/colcon_ws/src/ros2/rcl/rcl_action/src/rcl_action/goal_handle.c:95
```

witignite, 2021-03-16:
@fujitatomoya Hmm, that's strange. I rechecked again with *uncommented* [`goal_handle->publish_feedback(feedback);`](https://github.com/witignite/demos/blob/100fb00f7290714b260d4c12414f84265301551d/action_tutorials/action_tutorials_cpp/src/fibonacci_action_server.cpp#L120) and the  [`execute`](https://github.com/witignite/demos/blob/100fb00f7290714b260d4c12414f84265301551d/action_tutorials/action_tutorials_cpp/src/fibonacci_action_server.cpp#L97) is blocked.

KavenYau, 2021-05-02:
With both #1635 and #1641 were merged, is this issue fixed？

fujitatomoya, 2021-05-03:
confirmed with https://github.com/witignite/demos/blob/main/action_tutorials/action_tutorials_cpp/src/fibonacci_action_server.cpp, there is no coredump with canceling goal.

```bash
[Terminal - 1]
> ros2 run action_tutorials_cpp fibonacci_action_server
[INFO] [1620009868.893965965] [fibonacci_action_server]: Received goal request with order 10
[INFO] [1620009868.894794602] [fibonacci_action_server]: Executing goal
[INFO] [1620009868.895144634] [fibonacci_action_server]: Publish feedback
[INFO] [1620009869.895228669] [fibonacci_action_server]: Publish feedback
[INFO] [1620009870.895160211] [fibonacci_action_server]: Publish feedback
[INFO] [1620009871.895194236] [fibonacci_action_server]: Publish feedback
[INFO] [1620009872.895234427] [fibonacci_action_server]: Publish feedback
[INFO] [1620009873.150890128] [fibonacci_action_server]: Received request to cancel goal
[INFO] [1620009873.151048329] [fibonacci_action_server]: 		Cancel routine ... should take ~15 seconds
[INFO] [1620009873.151160620] [fibonacci_action_server]: 		cancel ... 0
[INFO] [1620009873.895149851] [fibonacci_action_server]: Publish feedback
[INFO] [1620009874.151364959] [fibonacci_action_server]: 		cancel ... 1
[INFO] [1620009874.895223791] [fibonacci_action_server]: Publish feedback
[INFO] [1620009875.151626042] [fibonacci_action_server]: 		cancel ... 2
[INFO] [1620009875.895230854] [fibonacci_action_server]: Publish feedback
[INFO] [1620009876.151922370] [fibonacci_action_server]: 		cancel ... 3
[INFO] [1620009876.895172636] [fibonacci_action_server]: Publish feedback
[INFO] [1620009877.152210830] [fibonacci_action_server]: 		cancel ... 4
[INFO] [1620009877.895817570] [fibonacci_action_server]: Goal succeeded
[INFO] [1620009878.152467187] [fibonacci_action_server]: 		cancel ... 5
[INFO] [1620009879.152733076] [fibonacci_action_server]: 		cancel ... 6
[INFO] [1620009880.153035109] [fibonacci_action_server]: 		cancel ... 7
[INFO] [1620009881.153296077] [fibonacci_action_server]: 		cancel ... 8
[INFO] [1620009882.153572034] [fibonacci_action_server]: 		cancel ... 9
[INFO] [1620009883.153863278] [fibonacci_action_server]: 		cancel ... 10
[INFO] [1620009884.154146435] [fibonacci_action_server]: 		cancel ... 11
[INFO] [1620009885.154423978] [fibonacci_action_server]: 		cancel ... 12
[INFO] [1620009886.154677246] [fibonacci_action_server]: 		cancel ... 13
[INFO] [1620009887.154921378] [fibonacci_action_server]: 		cancel ... 14
[INFO] [1620009888.155253001] [fibonacci_action_server]: 		Waited 15004 ms
[INFO] [1620009888.155405237] [fibonacci_action_server]: 		Cancel routine ... finished

[Terminal - 2]
> ros2 action send_goal /fibonacci action_tutorials_interfaces/action/Fibonacci "{order: 10}"
Waiting for an action server to become available...
Sending goal:
     order: 10

Goal accepted with ID: 13cfbaeafb0344a58c754530fa7a2169

^CCanceling goal...
Failed to cancel goal
```

@witignite it would be really appreciated if you do double check on this.

witignite, 2021-05-05:
Hi @fujitatomoya , I will try to double check as soon as I get back. Just curious if you use the current source code right? (Compile from source?)

fujitatomoya, 2021-05-05:
@witignite yes, mainline source code build. sorry i forgot to tell that.

witignite, 2021-05-13:
Hi @fujitatomoya, sorry to take this long. Confirmed. Using the current source code, with this line (`goal_handle->publish_feedback(feedback);`) commented out and not commented out, I now get the same behavior (expectedly), which is:
```bash
terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
  what():  goal_handle attempted invalid transition from state SUCCEEDED with event CANCEL_GOAL, at ~ros2_foxy/src/ros2/rcl/rcl_action/src/rcl_action/goal_handle.c:95
```

closed 2021-05-13 by witignite