ros2/rclcpp #1570: Rclcpp node crashes when using TinyXML inside service callback

coalman321, 2021-03-04:
## Bug report
- Operating System:
  - Ubuntu 20.04
- Installation type:
  - Binaries
- Version or commit hash:
  - Foxy
- DDS implementation:
  - default (FastRTPS?)
- Client library (if applicable):
  - rclcpp, TinyXML

#### Steps to reproduce issue
When using TinyXML inside a service to parse and manipulate an xml doc, the node will silently exit at the end of the service callback function. 

see relevant rclcpp based node in the link below. I have isolated the problem to being caused by a call to the LinkEndChild function when run against the parent XML document. It seems to be some behavior inside the function is causing the callback function to continue to run as expected, but crash the node on return of the callback function. Eg. the function continues to execute as expected, but after it returns the node crashes hard without an error.

Code file: https://gist.github.com/coalman321/a94d8201d7d3d1369dc073c8734a16dc  (Issue occurs at line 73)
Cmake file https://gist.github.com/coalman321/48995c5157f00bd0c7692eba1adc3667

#### Expected behavior
The node should continue to run as expected, and continue serving requests.

#### Actual behavior
The node silently exits with no indication as to why, and all callers get stuck waiting for a response.

#### Additional information
If I force the callback to return before line 73 is executed, the service will remain running as expected and continue to service new calls.

coalman321, 2021-03-04:
I found some more interesting behavior. If I convert the TinyXML documents, and other objects to pointers and manage them manually, the node will function as expected. 
The original behavior of the node exiting silently is still not clear, and is likely due to an object being de-referenced at the end of the function call when it had already been freed.

clalancette, 2021-03-04:
I took a look at this, just because it is weird and I was curious.

> The original behavior of the node exiting silently is still not clear, and is likely due to an object being de-referenced at the end of the function call when it had already been freed.

As far as I can tell, yes, this is the case.  I'm pretty sure the problem is because of the line:

```
TiXmlElement poseElem = TiXmlElement("pose");
```

That is creating the `poseElem` as a stack variable, and then feeding it into the document.  But when the destructor for `TiXmlDocument` runs, it expects to be able to free that memory from the heap.  Since `poseElem` is on the stack, that causes the crash.  Just changing that to a pointer fixes the issue for me.

In any case, this is not a bug in `rclcpp` or ROS 2, so I'm going to close this out.

closed 2021-03-04 by clalancette

coalman321, 2021-03-05:
That is a fair point, and I was able to get it running by converting it to pointers. 
From a relatively new user perspective though it still was not entirely clear what was going on. It would be helpful if the node could give an indication that there was an error while executing the callback function itself, and not just silently exiting. Not sure if this is feasible though, but just a user experience thought.

clalancette, 2021-03-05:
> That is a fair point, and I was able to get it running by converting it to pointers.
> From a relatively new user perspective though it still was not entirely clear what was going on. It would be [...]

Unfortunately, this is C++, and there is nothing we can do about it.  The code implemented in a callback can always cause a segfault, at which point all bets are off and the problem is completely out of our hands.

coalman321, 2021-03-05:
That is a good point. I have not used callbacks in c++ very much and was not sure if it would be possible.