ros2/rclcpp #1580: Inconsistent behaviour of spin_some after closing and reopening publisher

federicomassa, 2021-03-12:
## Bug report

**Required Info:**

- Operating System:
  - Pop!OS_20.04 LTS 
- Installation type:
  - docker
- Version or commit hash:
  - ros:foxy
- DDS implementation:
  - default
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

This bug (if it is a bug) is probably not exactly reproducible as is on every device. On my system, the steps are:

-  Insert the following code into a ros2 package

```cpp
#include <memory>

#include "rclcpp/rclcpp.hpp"
#include "std_msgs/msg/string.hpp"

using std::placeholders::_1;

class MinimalSubscriber : public rclcpp::Node
{
  public:
    MinimalSubscriber()
    : Node("minimal_subscriber")
    {
      subscription_ = this->create_subscription<std_msgs::msg::String>(
      "topic", 100, std::bind(&MinimalSubscriber::topic_callback, this, _1));
    }

  private:
    void topic_callback(const std_msgs::msg::String::SharedPtr msg) const
    {
      std::cout << "I heard: " << msg->data << std::endl;
    }
    rclcpp::Subscription<std_msgs::msg::String>::SharedPtr subscription_;
};

int main(int argc, char * argv[])
{
  rclcpp::init(argc, argv);
  rclcpp::Rate rate(1);

  auto node = std::make_shared<MinimalSubscriber>();

  while (rclcpp::ok()) {
    rclcpp::spin_some(node);
    rate.sleep();
  }

  rclcpp::shutdown();
  return 0;
}
```

- `colcon build`

On a terminal:
`ros2 run issue subscriber`

On a different terminal:
- `ros2 topic pub /topic std_msgs/String "data: ciao" -r 1`;
- Subscriber on the other terminal is receiving correctly;
- stop `ros2 topic pub` (CTRL+C)
- Play it again: `ros2 topic pub /topic std_msgs/String "data: test" -r 1`;

#### Expected behavior
Message is received again by the subscriber node on the second time `ros2 topic pub` is launched

#### Actual behavior
Message is **not** received the second time (and the next ones as well)

#### Additional information
I suspect this has something to do with the subscriber queue size and the node loop rate. At larger queue sizes (~100, 1Hz rate) this does not happen, nor with larger frequencies (at 100Hz, queue size 1). It does not seem to depend on the publishing rate.
**NB**: using spin() instead of spin_some() is not a solution for me.

fujitatomoya, 2021-03-14:
I do not think that this is bug, it is just about the frequency and QoS setting for publisher and subscription.

federicomassa, 2021-03-15:
Interesting. Can you elaborate on why you think it should be expected behavior to have two different results during the first and the second iteration of the publisher? I mean, wouldn't it be desireable that it deterministically works (or not work)?

clalancette, 2021-03-15:
For what it is worth, I can't reproduce on foxy on Ubuntu 20.04.  I can switch the publisher back and forth between the test strings, and the subscriber continues to receive them.  Indeed, if this didn't work, basically nothing in the ROS 2 ecosystem would work.

I'm not saying that there isn't a problem, just that there is probably something more involved going on here.  Possibly environment or docker related, but I'm not really sure.

@federicomassa Is there any chance you can run it on the real system, not in a docker container?  That at least reduces the number of variables here.

fujitatomoya, 2021-03-16:
> Can you elaborate on why you think it should be expected behavior to have two different results during the first and the second iteration of the publisher?

my bad, i didn't mention on that. i cannot so far reproduce your problem. I also use container ros:foxy and also checked with source build based on ubuntu:20.04 container. (My env is AMD Ryzen 9 3900X 12-Core Processor and host system is Ubuntu18.04)

federicomassa, 2021-03-25:
Thank you for your feedback! It's a pity that it's so hard to reproduce, I'll try to go deeper in the future and let you know if I find out something!

t-lin, 2023-01-22:
I've also been seeing this behaviour for the past year or so, and can confirm all of the observations in the original post.It's easiest to reproduce with a subscriber QoS depth of 1 and the rate/frequency of 1 Hz in the main loop. I've also found that adding `--once` to the publishing command (i.e. use `ros2 topic pub --once ...`) I can almost always get the listener to hang -- i.e. it won't receive the message and won't receive any future messages.