ros2/rclcpp #1585: Declaration of parameter of type std::vector<int> won't build

AugusteBourgois, 2021-03-19:
## Bug report

**Required Info:**

- Operating System: Ubuntu 20.04
- Installation type: binaries
- Version or commit hash: Foxy
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue
1. Create a ros2 package
```shell
ros2 pkg create --build-type ament_cmake vector_int_pkg
```
2. Create a simple node the declares a parameter of type `std::vector<int>` in file `src/test_vector_int.cpp`
```cpp
#include "rclcpp/rclcpp.hpp"

int main(int argc, char ** argv)
{
  rclcpp::init(argc, argv);

  std::shared_ptr<rclcpp::Node> node = rclcpp::Node::make_shared("vector_integer_parameters");

  node->declare_parameter("param", std::vector<int>{});

  rclcpp::spin(node);
  rclcpp::shutdown();
}
```
3. Adjust the CMakeLists.txt file
```cmake
cmake_minimum_required(VERSION 3.5)
project(vector_int_pkg)

if (NOT CMAKE_C_STANDARD)
    set(CMAKE_C_STANDARD 99)
endif ()

if (NOT CMAKE_CXX_STANDARD)
    set(CMAKE_CXX_STANDARD 14)
endif ()

if (CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
    add_compile_options(-Wall -Wextra -Wpedantic)
endif ()

find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)

add_executable(test_vector_int src/test_vector_int.cpp)
ament_target_dependencies(test_vector_int rclcpp)

ament_package()
```
4. Adjust the package.xml file
```xml
<?xml version="1.0"?>
<?xml-model href="http://download.ros.org/schema/package_format3.xsd" schematypens="http://www.w3.org/2001/XMLSchema"?>
<package format="3">
    <name>vector_int_pkg</name>
    <version>0.0.0</version>
    <description>TODO: Package description</description>
    <maintainer email="john@doe.com">John Doe</maintainer>
    <license>TODO: License declaration</license>

    <buildtool_depend>ament_cmake</buildtool_depend>

    <depend>rclcpp</depend>
[... 191 lines cut ...]
node->declare_parameter("param", std::vector<double>{});
// --------------------------------------
node->declare_parameter("param", rclcpp::ParameterValue(std::vector<int>{}));
```
Here are the lines 104 to 121 from the `/opt/ros/foxy/include/rclcpp/parameter_value.hpp` file
```cpp
  /// Construct a parameter value with type PARAMETER_BOOL_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<bool> & bool_array_value);
  /// Construct a parameter value with type PARAMETER_INTEGER_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<int> & int_array_value);
  /// Construct a parameter value with type PARAMETER_INTEGER_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<int64_t> & int_array_value);
  /// Construct a parameter value with type PARAMETER_DOUBLE_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<float> & double_array_value);
  /// Construct a parameter value with type PARAMETER_DOUBLE_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<double> & double_array_value);
  /// Construct a parameter value with type PARAMETER_STRING_ARRAY.
  RCLCPP_PUBLIC
  explicit ParameterValue(const std::vector<std::string> & string_array_value);
```

wjwwood, 2021-03-19:
Thanks for the thorough bug report! If anyone has time to investigate why this is, it would be appreciated.

I think what you're trying to do _should_ work, since it's initialization only, but we do only support `int64_t` and `double` under the hood, so if you use `vector<int>` or `vector<float>` it would be converted to `vector<int64_t>` and `vector<double>` respectively.

rpaaron, 2021-03-22:
so `vector<int>` is not `std::is_convertible` to `vector<int64_t>`, and so no valid class can be deduced from the template.
Is there a consensus on the correct fix?  If so, perhaps this would make a good first issue?

Without knowing too much about the background, I think I'd want it *not* to silently convert these vectors, and perhaps instead a template specialization to give a more explicit error would be helpful?  eg "hey, we dont support `vector<int>`, please use a `vector<int64_t>` instead"?

wjwwood, 2021-03-22:
> Is there a consensus on the correct fix? If so, perhaps this would make a good first issue?

Not sure, but it does seem like a small, self-contained issue that could make a good first issue.

> Without knowing too much about the background, I think I'd want it not to silently convert these vectors, and perhaps instead a template specialization to give a more explicit error would be helpful? eg "hey, we dont support vector<int>, please use a vector<int64_t> instead"?

Well, since it is initialization only, I don't actually see a problem with implicitly using the `vector<int>` to initialize a `vector<int64_t>`, since there's no risk of loss of precision and we were gonna allocate a new vector anyway (make a copy of the input).

Either way, an explicit specialization is the way to go, either with code to initialize the vector or give a better error.

jacobperron, 2021-04-09:
Interestingly, it looks like using the `NodeParametersInterface` directly works:

```c++
// These compile
parameters_interface_->declare_parameter(name, rclcpp::ParameterValue(std::vector<int>{}));
parameters_interface_->declare_parameter(name, rclcpp::ParameterValue(std::vector<int64_t>{}));
```

However, implicit conversion to `rclcpp::ParameterValue` does not work (I thought it would :shrug:):

```c++
// These don't compile
parameters_interface_->declare_parameter(name, std::vector<int>{});
parameters_interface_->declare_parameter(name, std::vector<int64_t>{});
```

closed 2021-07-07 by fujitatomoya