ros2/rclcpp #1586: Create load_parameters and delete_parameters methods on rclcpp::[A]syncParametersClient

BriceRenaudeau, 2021-03-19:
## Feature request

#### Feature description
It would be great to add these two methods to rclcpp::[A]syncParametersClient:

1. `load_parameters` to do the same as `ros2 param load [node] [yaml_file]` 
2. `delete_parameters` to do the same as `ros2 param delete [node] [parameters]`

#### Implementation considerations
Right now, I am using a system call to execute these two commands.

BriceRenaudeau, 2021-03-19:
To delete a parameter one can use :
```c++
// without value, it sets the parameter to NOT_SET, behave like delete
auto set_results = parameter_client->set_parameters({rclcpp::Parameter(param)});
```

wjwwood, 2021-03-19:
Makes sense to me, we only have them implemented in `rclpy`. Bonus points for considering if working it into `rcl` so `rclpy` and `rclcpp` can share the same implementation of these two functions makes sense.

BriceRenaudeau, 2021-03-24:
Linked issue #1029

sloretz, 2021-06-04:
Closing as this appears to have been resolved by #1596

closed 2021-06-04 by sloretz