ros2/rclcpp #1589: Add automatically_add_executor_with_node option on create_callback_group for rclcpp::LifecycleNode

BriceRenaudeau, 2021-03-22:
## Feature request

#### Feature description
When creating a callback_group for a lifecycleNode, the method `create_callback_group` uses the default value `true` for automatically_add_executor_with_node. It could be good to open this as a parameter like for rclcpp::Node.

#### Implementation considerations
It is possible to do it by using the lifecycle->get_base_node_interface.

wjwwood, 2021-03-22:
👍 

This would be a good first contribution for someone, but generally we would appreciate help with this change.

fujitatomoya, 2021-03-26:
@BriceRenaudeau 

thanks for the contribution 👍 
i am going to close this since it's been addressed with https://github.com/ros2/rclcpp/pull/1594.

CC: @wjwwood

closed 2021-03-26 by fujitatomoya